helium/router #556: CfList in US JoinAccept should not alternate

lthiery, 2021-12-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 - Console Device ID
 - Device dev eui / app eui
 - How often is device sending data
 - Region
 - Hotspots (names) used to transfer data

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

mikev, 2022-01-19:
Here is the current documentation for the CF List enable feature in the device profile:

>Enable Join-Accept CF List (applicable to US915 devices only)
>The Join-Accept CF List configures channels according to the LoRaWAN spec to use sub-band 2. Devices that have not correctly implemented the LoRaWAN spec may experience transfer issues when this setting is enabled.

> Enabled, the server will send a CF List with every other join.
> Disabled, the server will not send a CF List. The channel mask is still transmitted via ADR command.

This feature ONLY applies to US915 regions. Until very recently Helium mistakenly had reversed the CHMask bytes in the Join response.  See - https://github.com/helium/router/issues/535.  It is likely that the supposed implementation flaw of U.S. end-device software was actually caused by the reversed ChMask bytes.

mikev, 2022-01-21:
Lets also consider removing the CF List enable (U.S. only) feature completely.  My suspicion is this work-around was needed because the chmask was reversed, therefore when end-devices processed this, they started transmitting on the wrong channels.

jdgemm, 2022-01-26:
> Lets also consider removing the CF List enable (U.S. only) feature completely. My suspicion is this work-around was needed because the chmask was reversed, therefore when end-devices processed this, they started transmitting on the wrong channels.

@mikev let's discuss this during sync call, thanks

jdgemm, 2022-01-26:
Based on conversation, for future release we will make following changes:
- only send cflist for supported devices (1.1)
- remove alternating aspect of sending cflists
- change default from disabled to enabled

Oliv4945, 2022-01-27:
Hi,

I just want to highlight that LoRaWAN 1.0.3+ devices are also compatible with a US CFList

closed 2022-02-17 by macpie