helium/router #558: Joining problems

slavendam, 2021-12-21:
4 users from US today reported issues with Joining to network in last few days.
One user sent this:

> "Tried over 20 times with a known hotspot right next to me… Removed from the Helium console and re-added it again like yesterday and it worked the first time after that.
> Same keys"

Can this be Console issue after release of v2.1.8?

lthiery, 2021-12-21:
Can you provide any more details? Is CFList enabled? Do you see JoinAccept being sent down from Console? How is RSSI/SNR of JoinRequest? Does JoinRequest even show up on Console?

macpie, 2021-12-21:
Please use template https://github.com/helium/router/issues/new?assignees=macpie&labels=bug&template=bug_report.md&title=

slavendam, 2021-12-22:
> Can you provide any more details? Is CFList enabled? Do you see JoinAccept being sent down from Console? How is RSSI/SNR of JoinRequest? Does JoinRequest even show up on Console?

There is no CFList specifically enabled, just default settings.
There is no any record in Helium Console about JoinReq/JoinAcc.
Console Version 2.1.8
DevEUI 1D4A7D0000F59B4B
Region US915

lthiery, 2021-12-22:
Are you running your own Console? Or using helium.console.com? App/JoinEUI are also important when trying to trace logs.

slavendam, 2021-12-22:
> Are you running your own Console? Or using helium.console.com? App/JoinEUI are also important when trying to trace logs.

Using console.helium.com
AppEUI D84BCF434A2A33C8

lthiery, 2021-12-22:
If you're on helium console please send the UUID from the console. You can extract it from the URL when you are on decide view. We can check if we've seen anything from that device. But if it's not joining it's likely something to do with coverage. If the radio side was doing well we would at least see JoinAccept

slavendam, 2021-12-22:
> If you're on helium console please send the UUID from the console. 

be8efba9-f2df-4439-8a40-de9b9a811536
Problem was 18th Dec, ~10pm UTC

> But if it's not joining it's likely something to do with coverage. If the radio side was doing well we would at least see JoinAccept

It is not coverage problem. Hotspot is in the same room and it work as soon as device is readded.
Similar issue was like 1-2months ago when you couldn't see anything in console and then lot of JoinReq appeared in Console with delay.
So I believe this can be some issue with Console again, but you can take a look

slavendam, 2022-01-06:
Hi,
this became HUGE problem and should be solved ASAP!
We at GLAMOS had  not small unnecessary costs recently because of this issue. Users were reporting problems with device, we needed to spend money on production of new devices and shipping, but issue was on network side. 
This issue affects some businesses a lot and it is core part which disables all device functionalities. So I kindly ask you to investigate this ASAP, in Sprint 67, and release fix.

Situation number 1:
User from US region was able to join network on Dec 17th and 19th. After that he was not able to join even if area is covered with 10+ hotspots. There is no any records about Join Req in Helium Console. Nothing changed in device during that period.
I tried to readd device few times, didn't help. Today tried to move it to staging console, but still no luck.
Then we uploaded new keys to device, added it to production Console and it Joined from **first try**.
Keys that didn't work: 
DevEUI 1D4A7D0000F26AEB
AppEUI D84BCF46AE0B1770
ID from staging console where keys are still added a854179c-f3a1-438e-b136-6a7ab692f6a2

Situation 2:
User from Germany can't join, tried on 5 different locations, nothing worked. 50+ attempts to join.
Removing device from Console -> adding it again -> work from first try.

Situation 3:
Today network was down. In area of 150 hotspots I was not able to get any Join Req in Console, similar to previous situations. This was fixed in couple of hours. 

It seems that some keys become blocked or hidden inside of network. Like they reached some limit, hotspots can't see it and readding device clear that cache (but from Situation 1 that is not always case). 
I believe this issue started to happening with v2.1.8 of Console. 

Thanks

jdgemm, 2022-01-06:
hi, can you provide a device ID of a device that didn't work, but you haven't updated the keys and it's still trying to join? 

This will help us diagnose the issue. The staging keys you provided aren't helpful since there's no device attached to them.

slavendam, 2022-01-06:
@jdgemm  It is still added to Staging console.
Device ID **a854179c-f3a1-438e-b136-6a7ab692f6a2**

[image]

jdgemm, 2022-01-06:
yes, but as mentioned there needs to be a device attached to those keys so the team can trace join attempts to diagnose the issue

slavendam, 2022-01-06:
I just updated these keys to one device, but in EU region, and got Join Req in Staging console. Try to track join attempts 24h ago with this Device ID.

jdgemm, 2022-01-06:
hi thanks for helping us troubleshoot, but to make sure I understand on staging you used the same keys on a different device and it joined as expected?

slavendam, 2022-01-06:
These keys are originaly for US device. Tried to join with that - didn't work. Tried to join with other device and same keys in EU - didn't work. Tried again with US device - didn't work. Tried again in EU on other device - it worked

jdgemm, 2022-01-06:
Ok, thanks for the additional context, unfortunately once the device is working as expected the team's ability to diagnose is limited. Next time you try to join a device that doesn't work please just let it continue to attempt to join and share the device keys with us (specifically the device ID). That will enable the team to trace the issue and perform deeper diagnosis.

slavendam, 2022-01-11:
@jdgemm @macpie @lthiery Anyone now available?

One user has issue. He had 1 Join Request in Console when device didn't succeed to Join. 
He is now standing next to 2 hotspot. When he tries to Join he sees LED light blinking on hotspots, but nothing see in Console anymore.
Can you track in now before he tries to readd it to Console?

ID: c1caf7fa-c68c-427b-b2a4-6200101d013e
DevEUI: 1D4A7D0000E6146D
AppEUI: D84BCF480E997368

macpie, 2022-01-11:
> @jdgemm @macpie @lthiery Anyone now available?
>
> One user has issue. He had 1 Join Request in Console when device didn't succeed to Join. He is now standing next to 2 hotspot. When he tries to Join [...]

tracing now

slavendam, 2022-01-12:
Hotspot that couldn't join is 11xMPbEZ3AgifcUYYQwbHnwN735abhd82YRWKywxGmHXM6sQbhM

slavendam, 2022-01-12:
What can we conclude from these things?

macpie, 2022-01-12:
> What can we conclude from these things?

That specific hotspot might have issue transferring data

closed 2022-01-20 by macpie