helium/router #571: RxDelay is not respected for downlinks on existing device

Oliv4945, 2022-01-06:
**Describe the bug**
I increased `RxDelay` of an existing device from 1 to 5 seconds then rejoined.
The `RxDelay` is correct in the `JoinAccept`:

```
[2022-01-06 12:32:11]  DevAddr= 480003c7
[2022-01-06 12:32:11]  MacRx1DataRateOffset= 0
[2022-01-06 12:32:11]  MacRx2DataRate= 8
[2022-01-06 12:32:11]  MacRx1Delay= 5
```

But the downlinks are send with `RxDelay=1`: `2296841334 -2295841334 = 1000000us` 
```
Jan 06 11:32:12 raspberrypi helium_gateway[463]: uplink @2295841334 us, 905.10 MHz, Ok(DataRate(SF10, BW125)), snr: 13.8, rssi: -62, len: 23 from MacAddress(00:16:C0:01:FF:10:FC), module: gateway
Jan 06 11:32:13 raspberrypi helium_gateway[463]: rx1 downlink @2296841334 us, 926.90 MHz, DataRate(SF10, BW500), len: 22 via MacAddress(00:16:C0:01:FF:10:FC), module: gateway
```

**To Reproduce**
1. Change RxDelay on an existing profile from 1 to 5 second
2. Rejoin
3. Downlinks are not sent with the ne RxDelay while correct value is set in the `JoinAccept`

**Device Info (please complete the following information):**
 - Environment (Production, Staging) : [Production ](https://console.helium.com/devices/2af3778e-c44b-4c06-b714-796467d94a7a)(Helium hosted)
 - Console Device ID: 2af3778e-c44b-4c06-b714-796467d94a7a
 - How often is device sending data: Around 2 Hz
 - Region: US915
 - Hotspots (names) used to transfer data: great-plum-mallard (data only)

Device parameters:
[image]

Profile parameters:
[image]

Edit: I written this where I updated and existing device/profile, but creating a new device/profile does not work too, RxDelay is still 1 second

lthiery, 2022-01-06:
@Oliv4945 I ran into this before when my device was not ACK'ing the RxDelay configuration (ie: there should be a RXTimingSetupAns MAC Command in the first uplink). Can you check for this?

Oliv4945, 2022-01-06:
Hi @lthiery, I do not have the same understanding of the LoRaWAN specification: RXTimingSetupAns should be answered to RXTimingSetupReq so the NS knows that the device is now using the new RxDelay parameter.   
It makes no sense to send RXTimingSetupAns after the JoinAccept: the fact that the device is using the right session keys/devAddr is a proof itself that it got the JoinAccept. Furthermore, why confirming other values than RxDelay=1 when there is no default value?

lthiery, 2022-01-06:
@Oliv4945 I had the same train of thought after I replied to you. We're looking into it internally

dpezely, 2022-01-28:
Comment on PR [592](https://github.com/helium/router/pull/582#issuecomment-1024451245) confirms this is working on Staging.

closed 2022-01-28 by dpezely