helium/router #579: MQTT integration is not working

perone, 2022-01-11:
1. Are you using Helium hosted Console or running your own open source Console?
Hosted console

2. If you are on Helium hosted Console, which url are you currently on? 
N/A

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).
Organization name: Schema
Device ID: 9716512d-bb60-4b8e-afe9-771d07d2ece6

Please describe your issue:
MQTT integration doesn't work. I tried using hivemq cloud and the console always shows that the integration is not working, but it doesn't say what the problem is. I tried to connect manually using mosquitto client from my local computer to the cloud server and it is working fine, so it is clearly something on Helium side of the integration. I also tried with a custom mosquitto server but wasn't working too.

pvolodin, 2022-01-17:
I believe the problem is that MQTT over TLS isn't supported.

@perone Hive has (at least) two brokers, which one you've tried?

perone, 2022-01-17:
Maybe it was TLS because I just used TLS, but I wasn't even able to receive a TCP connection at my MQTT server, so it seems that there is a more fundamental issue beyond the TLS support. I used the `s1.eu.hivemq.cloud` hivemq broker. Are there any plans to support TLS ? Some brokers do not even allow non-TLS connections (i.e. hivemq). Thanks !

pvolodin, 2022-01-18:
@perone I bet if you try broker.hivemq.com:1883 (https://www.hivemq.com/public-mqtt-broker/) you'll get some success. But of course this broker cannot be used in production.

> Are there any plans to support TLS ?

I'm not a part of the Helium team so have no answer for this question. I experienced this (or very similar) issue some weeks ago so adding my two cents to the discussion/details. I finally switched to HTTP integration, but have MQTT working too would be great.

If you get success with Hive's public broker, then it probably will be a good idea to change the issue description.

jdgemm, 2022-01-18:
Thanks for reporting, we'll take a look at what's needed to support TLS.

takeabyte, 2022-01-18:
Its not only tls. Tried the adafruit broker via 1883 and had the same error.

perone, 2022-01-19:
@takeabyte would you mind sharing what was the issue ? In my case, I wasn't even receiving a connection on the server and I also tried the `mqtts://user:pass@host:port` w/ TLS without success.

takeabyte, 2022-01-19:
@perone i think i narrowed it down to just hivemq not working, not TLS in general. 
I signed up for Adafruit.io and it was able to connect via mqtts w/ TLS and push data to it. 
That lead me to believe the bug is more specific to hivemq - either the long domain name or the .cloud ending.
I also checked with both hivemq azure and aws containers, additionally made the user/pass combination simpler and as short as possible. Also no luck with hivemq.

perone, 2022-01-19:
Thanks @takeabyte ! I will try with other brokers to check too.

gilloud, 2022-01-23:
Similar issue without TLS (juste login/pass on a specific port) After sole messages i have an "intégration issue". No more mqtt connection retry.

macpie, 2022-01-24:
I have tested the [public HiveMQ broker](https://www.hivemq.com/public-mqtt-broker/) and I am receiving my data just fine.

I have tried their cloud version and it looks like SSL only works if `Whether a client verifies the server's certificate chain and host name` is disabled otherwise it seems that the certificate is expired.

takeabyte, 2022-01-25:
@macpie where can you change that setting?

macpie, 2022-01-25:
> @macpie where can you change that setting?

It is a setting on my mqtt client (MQTTX) it cannot be change in Router. it seems to be a security flaw on HiveMQ side that the certificate they are using is expired.

takeabyte, 2022-01-25:
https://community.hivemq.com/t/frequently-asked-questions/514

macpie, 2022-01-25:
Thanks, I will check with a newer version on our library to see if this potentially fix the issue.

takeabyte, 2022-01-25:
@macpie I tried to recreate your steps with MQTT Explorer and got the same Certification error until I downloaded added the ISRG Root X1.pem as Server Certificate (CA) within my MQTT Explorer Client Software. 
See https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

drberthol, 2022-02-04:
It is possible the Helium mqqt is set to use port 1883 ?  ( no way to overide this with the command line ?? )

macpie, 2022-02-12:
Seems like latest upgrade did not fix the specific problem for hivemq cloud all other TLS connection seems to be fine. I have attempted a lot of different fixes but nothing so far, I will rename this issue at it is specific to hivemq cloud.

Will attempt more next week.

sunburntcat, 2022-02-15:
My own setup has been piping to mosquitto MQTT broker (non-TLS) for over a month. It was working fine until February 10th, and now the Integration gives a warning and says there's an "integration error." 

Everything appears fine on my end. I tested with same mosquitto pub/sub topics and messages to the broker on various servers, and everything is working fine. No matter what test endpoints I give the Helium integration, it still gives the nondescript Integration Error messages.

Perhaps something was jostled in 2.2.2?

sunburntcat, 2022-02-16:
Thanks, macpie I created a new issue. It looks to be auto-assigned to you

jdgemm, 2022-03-01:
@perone we've made some updates can you confirm that hivemq is still experiencing issues? If so, what are they?

takeabyte, 2022-03-01:
hey @jdgemm 
I just tried it again. I still can't use my private hiveMQ cloud mqtts integration via 8883.
Integration request and response both from debug window read: 
  "category": "uplink",
  "description": "Error: not_connected"

  and it further reads down at debug request and response "status": "error"

tekkies, 2022-03-05:
I'm also seeing this issue.  Here's a more full debug trace, in case it helps.

I have a flow setup to post to both
- test.mosquitto.org:1883 : MQTT, unencrypted, unauthenticated
- hiveMQ: 8884: MQTT, encrypted, authenticated

test.mosquitto.org posts fine.

I have just setup a 3rd MQTT sink in my flow, but won't be in range until tomorrow.
- test.mosquitto.org:8885 : MQTT, encrypted, authenticated

`{
  "id": "9df6e5e3-0a07-402d-aa30-a4d277867eb9",
  "name": "HiveMQ",
  "status": "error"
}
{
  "body": {
    "app_eui": "6081F9EFD455B2BC",
    "dev_eui": "6081F9F868F6CE1B",
    "devaddr": "9A020048",
    "fcnt": 80,
    "hotspots": [
      {
        "channel": 3,
        "frequency": 867.0999755859375,
        "hold_time": 638,
        "id": "116CKALEtYV2NdNVpvWXotct9f6X824E3wqEm91xNwkbsG11kF5",
        "lat": 51.67405725509936,
        "long": -4.045851670815135,
        "name": "pet-mustard-panther",
        "reported_at": 1646430799981,
        "rssi": -99,
        "snr": 11.199999809265137,
        "spreading": "SF7BW125",
        "status": "success"
      }
    ],
    "id": "28046113-a1eb-42b7-9d76-92c56c46372e",
    "metadata": {
      "adr_allowed": false,
      "cf_list_enabled": false,
      "multi_buy": 1,
      "organization_id": "e5c7b4ed-b5bd-47c7-a195-0a57615dbae8",
      "rx_delay": 1,
      "rx_delay_actual": 1,
      "rx_delay_state": "rx_delay_established"
    },
    "name": "CubeCell A",
    "payload": "UQ==",
    "payload_size": 1,
    "port": 1,
    "raw_packet": "QJoCAEiAUAABrH5+A7w=",
    "replay": false,
    "reported_at": 1646430799981,
    "type": "uplink",
    "uuid": "65b9d0a1-2e96-4f65-9110-c831f0697c19"
  },
  "endpoint": "mqtts://helium02:(mypassword)@e3e51f51977c435fb45898a2ec5a1d16.s2.eu.hivemq.cloud:8884",
  "qos": 0,
  "topic": "tekkies.co.uk/Helium/28046113-a1eb-42b7-9d76-92c56c46372e/rx"
}
`

tekkies, 2022-03-06:
I ran a test against test.mosquitto.org using tls and credentials and this also failed (same as HiveMQ)
`mqtts://testUser:testPassword@test.mosquitto.org:8885`

guinp1n, 2022-04-20:
> I ran a test against test.mosquitto.org using tls and credentials and this also failed (same as HiveMQ)
> `mqtts://testUser:testPassword@test.mosquitto.org:8885`

For test.mosquitto.org:8885 are you sure you are using correct username / password (testUser:testPassword)? Documentation on the page https://test.mosquitto.org says:

> The authenticated listeners require a username / password:
> 
> **rw / readwrite** : read/write access to the # topic hierarchy
> **ro / readonly** : read only access to the # topic hierarchy
> **wo / writeonly** : write only access to the # topic hierarchy

robidog, 2022-04-26:
Chiming in here. Any additional info we can provide to help investigate the issue?

macpie, 2022-04-26:
I will check on the `test.mosquitto.org` server

macpie, 2022-04-26:
I have tested `test.mosquitto.org` on port `8885`, it seems that it had a self signed certificate which is considered NOT secure. I had to disabled SSL verification to be able to connect with any of my client, including in Router. We do not allow this in Router for security reason. See https://www.erlang.org/doc/man/ssl.html#type-client_verify_type for more info.

closed 2022-04-26 by macpie

robidog, 2022-04-30:
Thanks @macpie for following up. I think they use letsencrypt, not a self signed cert for their service. When testing  from Home Assistant MQTT client, I can successfully connect to HiveMQ when installing the root cert found at https://letsencrypt.org/certs/isrgrootx1.pem. Can this certificate be uploaded and stored in Helium Console?

Related discussion here: https://community.hivemq.com/t/helium-connection/1103/3