helium/router #586: NetID Transition

mikev, 2022-01-26:
**Introduction**

Helium poached it’s NetId without permission from the LoRa Alliance.  However Helium now has been assigned an official Helium NetID.  This issue designs and plans the migration from our poached NetID to our official NetID.

It will take time to migrate end-devices, because end-devices cache the NetID after a Join.  Rather than immediately replace the current NetID, the proposed plan will update the code to support multiple NetIDs.  This will allow the legacy Helium NetID to gradually phase out of existence until the official Helium NetID is everywhere.  However because NetIDs provide access to a limited range of addresses we desire a long-term design which supports multiple NetIDs.

**NetIDs**

Official
0x60002D  - (type 3) NwkAddr bits == 17
0xC00053 -  (type 6) NwkAddr bits == 10

Poached
$H - (type 0) NwkAddr bits = 25

**Migration Plan**

1. Deploy LoRaWAN library
2. Update & Deploy BlockChain Core
3. Assign official NetID to chain var
4. Update & Deploy Router
5. Update & Deploy Miner
6. Update & Deploy Rust light Gateway

**PRs**

Links to the NetID PRs

Code ported to Rust light gateway
https://github.com/helium/gateway-rs/pull/155

New LoRaWAN library repo
https://github.com/helium/erlang-lorawan/tree/develop

Update to the blockchain.  Executed from the Miner.  Handles the OUI routing subnet lookup.
https://github.com/helium/blockchain-core/pull/1067

Update to the Router.  Handles the Join request.
https://github.com/helium/router/pull/482

mfalkvidd, 2022-01-26:
Nice work. Will this require any special changes or manual work for self-hosted console/router?

Vagabond, 2022-01-26:
This should be seamless for routers that are running a release with this code loaded and understand the related chain vars.