helium/router #587: Device not able to join

disk91, 2022-01-26:
**Describe the bug**
A device is not able to join in different covered area. Later this device join the network normally.
What is specific is that usually when a such problem is hapeing the JOIN request is visible on the console but the JOIN ACCEPT is not received by the device. But in this case, even the JOIN REQUEST is not visible in the console.

I have no more details currently as this is happening during traveling and there is no log accessible

**To Reproduce**
impossible to detail 

**Device Info (please complete the following information):**
 - Environment (Production, Staging)  Production

 - Console Device ID -  25cd8397-44f2-45c6-9f9e-297e58033f67
 - Device dev eui / app eui - 6081F996C3BF6198
 - How often is device sending data - can be on every 30 seconds or even less. 
 - Region -  EU868
 - Hotspots (names) used to transfer data- unknown

jdgemm, 2022-01-26:
When you say: `this is happening during traveling `, you mean when the device is moving, correct?

macpie, 2022-02-14:
Probably an issue with local Hotspots, if nothing showed up in Console it most likely means that the hotspots were not able to send their data to Router.

closed 2022-02-14 by macpie