helium/router #590: Mustache template render failed

MacTelnetWork, 2022-01-28:
1. Are you using Helium hosted Console or running your own open source Console? Helium hosted

2. If you are on Helium hosted Console, which url are you currently on? https://console.helium.com/integrations/590de6a0-da26-4f47-b94d-2926acb9a097

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID 590de6a0-da26-4f47-b94d-2926acb9a097

Please describe your issue:

I have a single Integration for every Decoder Function with all fields from different sensors.

After the last console update tonight I get "mustache template render failed" in uplink request if the Integration requests fields that Function doesn't pass, before update I had null values and they were easy to manage,

Can we come back to how it was yesterday?

takeabyte, 2022-01-31:
i made the same observation. I use a custom Json Template where i group multiple sensors together yet having individual decoders for each sensor device type. 

`{
{{#decoded}}{{#payload}}
"sensora-door-open": {{DOOR_OPEN_STATUS}},
"sensora-alarm": {{ALARM}},
"sensora-battery-volts": {{BAT_V}},
"sensorb-event": "{{event}}",
"sensorb-power": "{{battery}}",
"sensorb-temperature": "{{temperature}}",
{{/payload}}{{/decoded}}
"name": "{{name}}",
"reported_at": {{reported_at}}
}
`

Running one or the other type of sensor it will give me the error "mustache template render failed" with Adafruit integration. Other custom HTTP integrations also fail in the same manor, just not throwing an error but not working as well. 
this configuration was running successful and untouched the last time around
2022/01/28 1:30:00AM CET

michaeldjeffrey, 2022-02-01:
@MacTelnetWork and @takeabyte we are reverting the template functionality to how it previously was. https://github.com/helium/router/pull/593

takeabyte, 2022-02-02:
@michaeldjeffrey still getting mustache error for some reason.

michaeldjeffrey, 2022-02-03:
@takeabyte sorry about that, there was a miscommunication on our end. 
The branch was merged, but didn't make it into the tag. 
We'll be tagging again soon.
I'll keep the issue open until this hits production.

closed 2022-02-10 by michaeldjeffrey