helium/router #592: OSS Console - Router wallet not returned, impossible to burn DCs

disk91, 2022-01-31:
Since update to version 2022.01.27 It not anymore possible to burn DCs for me
The api does not returns the router wallet ID and the the transaction can't be generated. (memo is correctly generated)
https://console.helium-iot.eu/api/data_credits/router_address return code 200
but body is:
```
{"address":""}
```

jdgemm, 2022-01-31:
hi, team is continuing to investigate..

disk91, 2022-02-02:
> Try kicking router container while console container is up.

It has no effect

disk91, 2022-02-02:
I also update to the last release. No effect

jdgemm, 2022-02-02:
The team is working on a patch to address this issue. Thanks for raising.

slavendam, 2022-02-03:
@disk91 Do you have problem that it:
1. never returns address 
or
2. sometimes returns (it can join and send some uplink), and later it does not return?

disk91, 2022-02-03:
Never returns ... at least all the time I have tested
You can do test on console.helium-iot.eu if you want ... it's open.

closed 2022-02-09 by jdgemm