helium/router #597: RxDelay set in the console is not propagated in the JoinAccept

Oliv4945, 2022-02-02:
**Describe the bug**
The value of the RxDelay set in the console is not sent to the device in the JoinAccept.

**To Reproduce**
1. Create a device profile with an RxDelay > 1. In my case CFList and ADR are set
2. Create a new device with this profile
3. Start the device an monitor the join accept
4. The join accept will be like: `E929A1 486532 0E080048 08 00 00FF0000000000000000000000000001` where RxDelay is set to the the [router default value of 0](https://github.com/helium/router/blob/8a103868831e99f72600ba34ef2f8ea2c3351e51/include/lorawan_vars.hrl#L10) (13th byte before the CFList).

**Expected behavior**
The value set in the console is sent in the joinAccept

**Device Info (please complete the following information):**
 - Environment (Production, Staging): both
 - Console Device ID: 953f8b66-9f13-4df4-bbc4-881ccc4f8aa0 (Prod) - 6dc0f3fa-cb37-4fa9-bb17-e19b80676fe1 (US)
 - How often is device sending data : Every 15 min for the device in prod. Once in a while for the device on staging
 - Region: EU868 (Prod) - US915 (Staging)
 - Hotspots (names) used to transfer data:  real-smoke-nightingale (Prod) - great-plum-mallard (Staging)

**Additional context**
The RxDelay was correctly set when I opened issue #571 and I belive it was working Moday on staging when I commented on #582 but I did not kept the logs to check again

Oliv4945, 2022-02-02:
Additional comment: As the console [only allows the range [1;15]](https://github.com/helium/console/blob/c6912b521926455ee0d4172dbd8e8183bdbae186/assets/js/components/config_profiles/ConfigProfileSettings.jsx#L149) and the default value of 0 is transmitted, I wonder if RxDelay is correctly transmitted by the console to the router, please tell me if I should open a console issue instead

dpezely, 2022-02-02:
@Oliv4945 we're investigating...

Oliv4945, 2022-02-04:
@dpezely: No MAC command downlinks are triggered by either:
* Changing RxDelay in the profile
* Selecting another profile

Tested on staging with device id `6dc0f3fa-cb37-4fa9-bb17-e19b80676fe1`

closed 2022-02-08 by macpie

ogimenez-smtc, 2022-02-09:
Hi,

The RxDelay is working for me in staging, thanks. Any ETA for prod deployment?

DeviceId: 
[image]

```
20714B3648653208 080048 08 05 00FF0000000000000000000000000001
DevAddr= 48000808
MacRx1DataRateOffset= 0
MacRx2DataRate= 8
MacRx1Delay= 5
```

Then changing it thanks to `RXTimingSetupReq` also works:

```
  *************************************
  * Receive a Valid downlink RX1 for Hook Id = 3
  *************************************

Cmd rx_timing_setup_parser = 4
Downlink frame  - (1 bytes): 00
DL Port = 20 , DL SNR = 40 , DL RSSI = 58 , DL Freq = 927500000 , DL DR = 13 , DL Fpending Bit = 0
```
But I am surprised that we have to create an application downlink for that, is it expected that the NS is not triggering the downlink by itself?

PR #600 (2022-02-08): Join-Accept implies `rx_timing_setup_ans` for `RXDelay`
LoRaWAN Join-Accept implies `rx_timing_setup_ans` because `RXDelay` was in the Join-Accept.

- 6.2.6 Join-Accept frame, which includes RXDelay
- This should resolve #597
- Starting this PR as a Draft because the test framework may also need to be corrected

LoRaWAN Join-Accept implies `rx_timing_setup_ans` because `RXDelay` was in the Join-Accept frame, but router's Join logic doesn't explicitly set that answer.

That is, the ACK from the device is equivalent to `rx_timing_setup_ans` in FOpts in a frame (after join-accept), but FOpts are  [... cut]