helium/router #611: Advances JSON Message Template not including DC Balance

takeabyte, 2022-02-10:
When trying to pass the DC Balance to an integration Endpoint it will show the correct amount of DC in the Output Preview Window  next to the Template Body, however it will not be sent out to the Endpoint, but instead show the Inverted section, which means the key doesn't exist, is false, or is an empty list.
 

```
{{#dc}}
"dc_balance": "{{balance}}",
"nonce": "{{nonce}}",
{{/dc}}
{{^dc}}
  "No DC Found",
{{/dc}}
```

closed 2022-03-18 by macpie

takeabyte, 2022-03-24:
unfortunately with current console v.2.2.5 DC Balance still won't show up on the Endpoint. 
Expected output should be
`"category": "uplink",
    "data": {
      "dc": {
        "balance": 914118,
        "nonce": 2,`

Advanced JSON Template in Use: 
`{
"reported_at": {{reported_at}},
{{#dc}}
"dc_balance": "{{balance}}",
"nonce": "{{nonce}}"
{{/dc}}
}`

Result coming back from integration:
`{
"reported_at": 1648160197185,
"dc_balance": ,
"nonce": 2
}`

michaeldjeffrey, 2022-03-24:
> Advanced JSON Template in Use: `{ "reported_at": {{reported_at}}, {{#dc}} "dc_balance": "{{balance}}", "nonce": "{{nonce}}" {{/dc}} }`

Can you try with the key `{{ dc_balance }}`?
https://github.com/helium/router/blob/master/src/device/router_device_channels_worker.erl#L667-L670

macpie, 2022-03-24:
The current key is `dc_balance`, mistake on our part we will update it to `balance` asap.

takeabyte, 2022-03-25:
yes. that is working for now. good you will change it soon, also to be in line with the Advanced JSON testoutput.

macpie, 2022-03-25:
@takeabyte  This should be corrected in `2022.03.16.2` and was just pushed

takeabyte, 2022-03-25:
yes. all good now. Thanks!