helium/router #617: Enhancement - Set higher default RxDelay

Oliv4945, 2022-02-14:
Now that RxDelay is working as expected it would be great to have a default value working for most setup. From the few gateways I monitor a value of 4-5 sec should work most of the time, but I have a gateway with GSM backhaul where 7 would be recommended. Do you have statistics of the `hold_time` on all gateways connected to your network server?

macpie, 2022-02-14:
We are improving communication with hotspots so until then we probably won't make decision on a "network wide" rxdelay.

macpie, 2022-05-23:
> Do you have statistics of the hold_time on all gateways connected to your network server?

Each Router has its own stats on this. So I cannot speak for the entire Helium network but in our case the percentiles are:

- 95% of packets are delivered under 3.5s
- 90% of packets are delivered under 2s
- Anything else is way bellow the 1s threshold
_(Note that these numbers are not only for uplink that need an ack or downlink but ALL uplinks)_

Some hotspots are still obviously outliers but it seems to be minority. I am still hoping that [gateway_rs](https://github.com/helium/gateway-rs) will improve this.

If this becomes more of an issue later, we could add a Router level option to change `rx_delay` so that each Router operator could change their own.

closed 2022-05-23 by macpie