helium/router #620: Data credits are not burned correctly

slavendam, 2022-02-14:
In 16 Uplink messages Console showed total of 537 packets (number of hotspots). Only 298 DCs were burned.

jdgemm, 2022-03-01:
Can you provide more details? The way this is currently worded has confused the team.

slavendam, 2022-03-01:
If we receive 20 packets in 1 message (1 uplink record), 20DCs should be burned. 
Instead of that, we really receive 20 packets in Console, but in avg only 55% (~11DCs) of that number DCs are burned.

mrx23dot, 2022-03-05:
Another one, has anyone noticed that balance is reduced for every gateway receiving uplink, and not just once for uplink frame?

Is this expected? Why would have to user pay more if the network grows?

macpie, 2022-03-05:
> Another one, has anyone noticed that balance is reduced for every gateway receiving uplink, and not just once for uplink frame?
> 
> Is this expected? Why would have to user pay more if the network grows?

Yes this is expected, you are paying for each copy of packet so that each hotspot can then be rewarded. If you only want one copy of each frame you can set it that way via  the "multi buy feature".

macpie, 2022-03-17:
After some investigation, I could not find any major discrepancy in balances. I think the DC tracking mechanism is due for an update. We will use this issue as a place holder for further improvements

macpie, 2022-05-23:
This was check at length no issue found.

closed 2022-05-23 by macpie