helium/router #621: Huge amount of late packets

slavendam, 2022-02-14:
In area with 70 hotspots we are getting 20-30 late packets.
In area with 500 hotspots, number of late packets goes up to 200.

We came to this numbers by sending 200+ messages and analysing which hotspots received messages everytime. In rare occasions we receive message without late packets (3/200 messages).

Can increasing wait time (ROUTER_FRAME_TIMEOUT) temporary help fixing this issue until you release new solution?
https://github.com/helium/router/blob/590ea77de55fea67c32673f38e7633c5092bef9d/.env-template#L28

jdgemm, 2022-02-14:
At this point perhaps the best thing for your team and business is to host your own Console/Router. This will allow your team to make the configuration changes you want specific for your environment. In addition, hosting in your region will eliminate any potential issues due to latency.

closed 2022-02-15 by macpie

slavendam, 2022-02-15:
Late packets are real issue for many users. Not just for me, and not just in one region (EU, US, AU) and they are not connected with device but with Console.

jdgemm, 2022-02-15:
Sorry, forgot to add, as mentioned as a comment to another github issue there is an broader Hotspot update that the team believes will help stabilize Hotspot data transfer because hotspots will no longer need to sync with the blockchain and be free to transfer packets all the time.

That's why we're closing this issue for now. But if the issue persists _after_ the light Hotspot release to the broader Hotspot population, and you still decide not to host your own Console/Router (still highly recommended), this issue can be re-considered.

This update will be the first phase of light Hotspots and is anticipated to be released "soon". Follow progress here: https://docs.helium.com/mine-hnt/light-hotspots/#current-progress-on-testnet.

jamesl-dm, 2022-06-08:
I'm still seeing hold times of 850 ms from Australia - roughly 4 times my ping to the US.