helium/router #623: Previously-used Custom MQTT integration now fails

sunburntcat, 2022-02-16:
**Bug Description**
Previously-used Custom MQTT integration fails with "Integration was unsuccessful." The failures began happening on February 10 ... the same date the Helium v2.2.2 Release Notes were published.

**To Reproduce**
Run MQTT broker without TLS from public accessible endpoint. I am running Mosquitto version 2.0.14 on https://kandaweather-mainnet.ddns.net:1883 .

In Helium console, created custom MQTT integration with endpoint:

Endpoint:  mqtt://kandaweather-mainnet.ddns.net
Uplink Topic: helium/dclimateiot/rx
Downlink Topic: helium/dclimateiot/tx

And then connected Flow as:   device -> decoder function -> Custom MQTT integration

The Decoder functions are well-tested. No changes were made to the flows or data processing server around the time of the Integration failures. Multiple devices fall under this Label, and all are failing to have their data sent over MQTT.

**Expected behavior**
The MQTT integration (and respective devices, decoding functions, etc) was working properly throughout the end of January and beginning of February. This was evidenced by the weather data being handled correctly after processing.

It was expected to also be working since February 10, but has been failing.

**Device Info (please complete the following information):**
 - Console Device ID:   ablwxnvl3a1x, dxujgds3gkzy, ... others
 - Device dev eui / app eui. :  
       App EUI:      MSB.  0x60, 0x81, 0xF9, 0x14, 0x7B, 0x09, 0xCE, 0x09
 - How often is device sending data : 15 minutes
 - Region:  U.S. centered around 915MHz
 - Hotspots (names) :  >3 public hotspots located in Dallas area (actual receiving hotspots change)

**Additional context**
The event_debug.json file downloaded in the "Devices" section only says 

"description": "Error: not_connected",

[image]

sunburntcat, 2022-02-16:
As of now, I have also utilized a different public broker

mqtt://broker.hivemq.com

And the same issue occurs

macpie, 2022-02-16:
> **Bug Description** Previously-used Custom MQTT integration fails with "Integration was unsuccessful." The failures began happening on February 10 ... the same date the Helium v2.2.2 Release Notes were published.
> 
> **To Reproduce** Run MQTT broker without TLS from public accessible endpoint. I am running Mosquitto version 2.0.14 on kandaweather-mainnet.ddns.net:1883 .
> 
> In Helium console, created custom MQTT integration with endpoint:
> 
> Endpoint: mqtt://kandaweather-mainnet.ddns.net Uplink Topic: helium/dclimateiot/rx Downlink Topic: helium/dclimateiot/tx
> 
> And then connected Flow as: device -> decoder function -> Custom MQTT integration
> 
> The Decoder functions are well-tested. No changes were made to the flows or data processing server around the time of the Integration failures. Multiple devices fall under this Label, and all are failing to have their data sent over MQTT.
> 
> **Expected behavior** The MQTT integration (and respective devices, decoding functions, etc) was working properly throughout the end of January and beginning of February. This was evidenced by the weather data being handled correctly after processing.
> 
> It was expected to also be working since February 10, but has been failing.
> 
> **Device Info (please complete the following information):**
> 
>     * Console Device ID:   ablwxnvl3a1x, dxujgds3gkzy, ... others
> 
>     * Device dev eui / app eui. :
>       App EUI:      MSB.  0x60, 0x81, 0xF9, 0x14, 0x7B, 0x09, 0xCE, 0x09
> 
>     * How often is device sending data : 15 minutes
> 
>     * Region:  U.S. centered around 915MHz
> 
>     * Hotspots (names) :  >3 public hotspots located in Dallas area (actual receiving hotspots change)
> 
> 
> **Additional context** The event_debug.json file downloaded in the "Devices" section only says
> 
> "description": "Error: not_connected",
> 
> [image]

Same issue here you are missing the port, please use `mqtt://kandaweather-mainnet.ddns.net:1883`

macpie, 2022-02-16:
NOTE: to avoid missing port in the future, Router will now default it to `1883`. See PR https://github.com/helium/router/pull/624

closed 2022-02-16 by macpie