helium/router #625: Azure IoT Hub "Error Not Connected"

DavidDeeds, 2022-02-16:
**Describe the bug**
Azure IoT Hub is not connecting, even after closure of bugs  #613 and #619 

The integration message is as follows:
{
  "category": "uplink",
  "description": "Error: not_connected",
  "device_id": "76549ca0-1b54-4e10-9ba3-0416660aa456",
  "device_name": "elsys_s-2034c",
  "id": "c360017a-2169-430d-b205-acea10d35b7e",
  "inserted_at": "2022-02-16T23:42:32",
  "organization_id": "a8dfd2a1-e0a6-46a5-8a72-8da583bec2ce",
  "reported_at": "1645054952693",
  "reported_at_naive": "2022-02-16T23:42:32",
  "router_uuid": "7aca0772-f741-4d2d-a5be-8bf2bb8ed570",
  "sub_category": "uplink_integration_req",
  "updated_at": "2022-02-16T23:42:32"
}

**To Reproduce**
Configure integration for Azure IoT hub.

**Expected behavior**
Integration to work as expected prior to console v2.2.2 release

**Device Info (please complete the following information):**
 - Prod 
 - 76549ca0-1b54-4e10-9ba3-0416660aa456
 - All devices
 - Every few minutes
 - Au
 - Alert Pear Tiger

*NOTE: do not post your device `app_key`*

**Additional context**
Have tried recreating integrations, confirmed Azure keys, regenerated azure keys, verified Azure IoT Hub works with another platform etc.

**Screenshots**
[image]

closed 2022-02-17 by macpie

macpie, 2022-02-18:
This is deployed to [staging-console.helium.wtf](https://staging-console.helium.wtf/)

EnrSal, 2022-02-25:
I am still getting "error not connected" on IoT hub integration.

DavidDeeds, 2022-02-25:
> I am still getting "error not connected" on IoT hub integration.

Same *sigh*... Just over two weeks of no data making it's way to dashboards... I've tried creating hubs, integrations etc. etc. etc. all to no avail.. Very very disappointing.

michaeldjeffrey, 2022-02-25:
@EnrSal and @DavidDeeds can you create a new issue with your device and channel information so I can look into it?
On my end it helps if you have the channel connected, even if it's failing.

maxnocker, 2022-05-15:
I am again getting an error status for integration requeset and integration response for all my devices. I changed for the devices and in the azure portal

EnrSal, 2022-05-16:
> I am again getting an error status for integration request and integration response for all my devices. I changed for the devices and in the azure portal

Azure IoT hub is also giving me an integration error. It has been working perfectly for 2 months since the last fix.

JasonTimmins, 2022-05-17:
Admittedly, I'm new to this but I'm seeing connection issues too. It feels like I'm doing something wrong but I'm pretty sure I'm not, I've made Azure IoT Hubs before and connected them to devices using TTN and directly.

[image]