helium/router #633: LinkAdrReq can set to DR4 on Fixed Channel Plans when CFList Disabled

lthiery, 2022-02-24:
In Fixed Channel Plans (such as US915 and AU915), when CFList is disabled, ChMask selecting the Helium Network's subband is communicated via LinkAdrReq. We use the multiple LinkAdrReq approach as described here
[image]

When a JoinRequest and the subsequent Uplink uses 500 MHz, it can compel Router to provide a LinkAdrReq with DR set to 4:
[image]

Since Channel 65 is disabled in the ChMask, setting DR to 4 is incompatible; DR4 is not valid on CH8-15.

Note, a JoinRequest and subsequent Uplink using 125 MHz does not compel DR to be set to 4:
[image]

Therefore, it appears that DR configuration in the LinkAdrReq is configured in response to the DR of the Uplink. This logic may be appropriate but must not exceed DR3. Another approach would be to consider enabling CH65 which would make DR4 possible.

mikev, 2022-02-24:
@lthiery - Can you link to a full json and csv packet trace?  Ideally the trace will include the join request and include a portion where FCnt starts at zero.

lthiery, 2022-02-24:
[dr4-test.csv](https://github.com/helium/router/files/8136380/dr4-test.csv)

You can replay the above with the latest release of [LoRaWAN Sniffer utility](https://github.com/helium/lorawan-sniffer/releases/tag/v0.2.4).

It looks like this:
```
JoinRequest	914.2 MHz 	DataRate(SF8, BW500)	RSSI: -112	LSNR: 5.5
	  AppEui: 35BEED137ACD344B DevEui: 653D0F09DDED4DB0 DevNonce: 9AD2
JoinAccept	927.5 MHz 	DataRate(SF7, BW500)
	AppNonce: 15664F NetId: 326548 DevAddr: 48000287
	DL Settings: DLSettings(8) RxDelay: 1
	Newskey: AES128([FF, 52, C8, 73, 26, 99, 9A, A6, DE, DF, 76, 59, C8, 4, 35, 5A])
	Appskey: AES128([D1, 60, 63, 9F, DF, A3, 2E, 8A, AB, 2E, A1, C6, 4F, 1E, 2A, CB])
DataUp	906.2 MHz 	DataRate(SF8, BW500)	RSSI: -112	LSNR: 5.5
	DevAddr: 48000287, FCtrl(0, true), FCnt x0, FPort 116
	DevEui[-4..]: 4DB0, Data: [14, ea, 21, 9]
DataDown	924.5 MHz 	DataRate(SF7, BW500)
	DevAddr: 48000287, FCtrl(42, false), FCnt x0, No FPort
	LinkAdrReqPayload(DR(4), TxPower(0), ChannelMask([0, 0]), Redundancy(70))
	LinkAdrReqPayload(DR(4), TxPower(0), ChannelMask([0, ff]), Redundancy(0))
	DevEui[-4..]: 4DB0
DataUp	907.8 MHz 	DataRate(SF8, BW500)	RSSI: -112	LSNR: 5.5
	DevAddr: 48000287, FCtrl(4, true), FCnt x1, FPort 183
	LinkADRAns(LinkADRAnsPayload([7]))
	LinkADRAns(LinkADRAnsPayload([7]))
	DevEui[-4..]: 4DB0, Data: [50, 4d, 9d, 86]
DataDown	925.1 MHz 	DataRate(SF7, BW500)
	DevAddr: 48000287, FCtrl(32, false), FCnt x1, No FPort
	DevEui[-4..]: 4DB0
DataUp	914.2 MHz 	DataRate(SF8, BW500)	RSSI: -112	LSNR: 5.5
	DevAddr: 48000287, FCtrl(0, true), FCnt x2, FPort 186
	DevEui[-4..]: 4DB0, Data: [c8, 81, 39, 21]
DataDown	927.5 MHz 	DataRate(SF7, BW500)
	DevAddr: 48000287, FCtrl(32, false), FCnt x2, No FPort
	DevEui[-4..]: 4DB0
```

mikev, 2022-02-25:
Confirmed the FOpts decode for Downlink-0 and Uplink-1

FCnt=0 CtrlBits=10 Downlink FOpts = [{link_adr_req,4,0,0,7,0},{link_adr_req,4,0,65280,0,0}]
  ChMaskCntl = 7
  ChMask = 0x0000
  DataRate = 4
  TXPower = 0
  NbTrans = 0
  ChMaskCntl = 0
  ChMask = 0xFF00
  DataRate = 4
  TXPower = 0
  NbTrans = 0
FCnt=1 CtrlBits=0 Uplink FOpts = [{link_adr_ans,1,1,1},{link_adr_ans,1,1,1}]

mikev, 2022-02-26:
[image]

[image]

Observations:

The device sends a JoinReq on 914.2 MHz @ DataRate(SF8, BW500) which translates to Ch71.
The next Uplink is 906.2 MHz @ DataRate(SF8, BW500)

Per the spec: "The Data Rate specified in the command need not be valid for channels specified in the ChMask, as it governs the global operational state of the end-device."  Examining the Router code I believe we are sending on DR4 because the device sent on DR4 and we initially mirror the receiving DR setting.  Further, the spec does not require conformance to our ChMask.

Could this be the source of the problem on the end-device?

pvolodin, 2022-02-26:
@lthiery 

>Another approach would be to consider enabling CH65 which would make DR4 possible.

Then one day you'll have a lot of devices using/sharing the single channel. It would be better to avoid this situation.

pvolodin, 2022-02-26:
@lthiery 

>It looks like this:

It looks like the channels which aren't supported by Helium (71 and 66) were used?

mikev, 2022-02-27:
Investigation Observations:

The LinkAdrReq would normally not be sent as the 2nd Downlink following the JoinAccept, however Helium's code is currently implemented this way as a work-around to a previous issue.  This work-around is due to be removed in the next release, See - https://github.com/helium/router/issues/556.  Therefore after #556 is deployed this may not be an issue.  Once ADR has a history of 20 packets it is known to work correctly.

Having said this, it is still worth considering if this is a bug.  The spec is clear:  "The Data Rate specified in the command need not be valid for channels specified in the ChMask, as it governs the global operational state of the end-device."  From what I gather the customer is testing their Gateway and has hard-coded various test scenarios so their testing may not be representative of a valid end-device (Need to confirm this).  The spec also states: "An end-device SHALL acknowledge the successful receipt of a command that specifies a higher TX power than it is capable of using. In that case, the end-device SHALL operate at its maximum possible power."  So again this indicates that a proper client should ignore settings it cannot use.

Another observation: this LinkAdrReq was intended to set the ChMask and it didn't need to set the DR or TxPower.  The spec allows The value 0xF (decimal 15) of either DataRate or TXPower means that the end-device ignore that field and keep the current parameter values".  So if the code's intent is only to set the ChMask then DataRate and TXPower should both be set to 0xF.

Recommendations:
1. If it is simple to implement then set DataRate and TXPower to 0xF, indicating to the client that only the ChMask is relevant.
2. Wait for #556 to be deployed.  It should be soon; ETA is this Sprint.

lthiery, 2022-02-27:
> The spec allows The value 0xF (decimal 15) of either DataRate or TXPower means that the end-device ignore that field and keep the current parameter values". So if the code's intent is only to set the ChMask then DataRate and TXPower should both be set to 0xF.

Ah that's a fantastic find!

pvolodin, 2022-02-27:
> > The spec allows The value 0xF (decimal 15) of either DataRate or TXPower means that the end-device ignore that field and keep the current parameter values". So if the code's intent is only to set the ChMask then DataRate and TXPower should both be set to 0xF.
> 
> Ah that's a fantastic find

From the regional parameters doc:
"DR15 and TXPower15 are defined in the LinkADRReq MAC command of the LoRaWAN® **1.0.4** and subsequent specifications and were previously RFU"

> But it is very odd that the end-device is using a non-DR0 Uplink setting during the Join. This indicates some issue with the end-device LoRaWAN library.

From the regional parameters doc:
"If using the over-the-air activation procedure, the end-device SHALL transmit the Join578 Request message on random 125 kHz channels amongst the 64 125kHz channels defined using DR0 and on 500 kHz channels amongst the 8 500kHz channels defined using **DR4**. The end-device SHALL change channels for every transmission.

For rapid network acquisition in mixed gateway channel plan environments, the device SHOULD follow a random channel selection sequence which efficiently probes the octet groups of eight 125 kHz channels **followed by probing one 500 kHz channel** each pass. Each consecutive pass SHOULD NOT select a channel that was used in a previous pass, until a Join-request is transmitted on every channel, after which the entire process can restart. "

pvolodin, 2022-02-27:
@mikev 

> The spec is clear: "The Data Rate specified in the command need not be valid for channels specified in the ChMask, as it governs the global operational state of the end-device."

For the US region, this is correct only for ChMaskCnt 6 (all 125 KHx channels ON) and 7 (all 125 KHx channels OFF) as they set the state of the channels not included in the channel mask. (== governs the global operational state of the end-device). Actually, this can be correct only for ChMaskCnt 6, as if you set DR0-DR3 together with ChMaskCnt 7, the device shall set DataRateACK bit of the Status field in the LinkADRAns to 0 ("The data rate requested is unknown to the end-device or is not possible given the channel mask provided (not supported by any of the enabled channels). The command was discarded and the end-device state was not changed")

pvolodin, 2022-03-04:
> @lthiery
> 
> > Another approach would be to consider enabling CH65 which would make DR4 possible.
> 
> Then one day you'll have a lot of devices using/sharing the single channel. It would be better to avoid this situation.

#641 seems to be an example

lthiery, 2022-03-04:
> @lthiery
> 
> > Another approach would be to consider enabling CH65 which would make DR4 possible.
> 
> Then one day you'll have a lot of devices using/sharing the single channel. It would be better to avoid this situation.

I wonder if the implementation of DR4 on a US end-device should be that they _may_ randomly pick the higher bands and use DR4 but if they randomly pick a lower band, they use DR3. I'll have to read the spec to see how it's phrased.

pvolodin, 2022-03-04:
>I wonder if the implementation of DR4 on a US end-device should be that they may randomly pick the higher bands and use DR4 but if they randomly pick a lower band, they use DR3. I'll have to read the spec to see how it's phrased.

It surely is a good idea to check with the docs, but there is a couple of details to take into account:

1) Receiver sensitivity depends on DR, max. allowed TX power depends on the number of available channels, and link budget depends on receiver sensitivity and TX power. So with 8+1 channel plan there's a (minor) chance that the hotspots are able to receive the packets from a device on DR4 or DR0 and unable to receive the packets from that device on DR3.

2) There's no hope all the devices in the network support LoRaWAN 1.0.4. Actually, I'm almost sure most of them are 1.0.2 or even lower. So you'll need to review all the specs.

Actaully, this issue demonstrates why DR15 and TXPower15 were introduced in the latest versions of the spec and RP.

closed 2022-03-21 by mikev