helium/router #634: Azure IoT Hub "Error Not Connected"

DavidDeeds, 2022-02-25:
**Describe the bug**
Azure IoT Hub is not connecting in Integrations. 

The integration message is as follows:
{
  "category": "uplink",
  "description": "Error: not_connected",
  "device_id": "e429da00-b945-4a6b-b376-225ad295ff34",
  "device_name": "netvox_sn201600012",
  "id": "f7218861-ccd1-4196-b323-3fb83cb59ae5",
  "inserted_at": "2022-02-25T09:33:34",
  "organization_id": "a8dfd2a1-e0a6-46a5-8a72-8da583bec2ce",
  "reported_at": "1645781614052",
  "reported_at_naive": "2022-02-25T09:33:34",
  "router_uuid": "d5fa956c-2b93-4711-8cbe-5aa78f196397",
  "sub_category": "uplink_integration_req",
  "updated_at": "2022-02-25T09:33:34"
}

**To Reproduce**
Configure integration for Azure IoT hub. See pictures for my flow.

**Expected behavior**
Integration to work as expected prior to console v2.2.2 release

**Device Info (please complete the following information):**
-Prod
-76549ca0-1b54-4e10-9ba3-0416660aa456
-All devices
-Every few minutes
-Au
-Alert Pear Tiger

**Additional context**
Raising this new bug report as mentioned by @michaeldjeffrey here https://github.com/helium/router/issues/625 

**Screenshots**
[image]

[image]

EnrSal, 2022-02-25:
The IoT integration is working again! What occurred?

michaeldjeffrey, 2022-02-25:
@EnrSal glad to hear it!
@DavidDeeds could you check your integration again and verify things are working for you as well?

There is/was an ongoing issue with our container manager (quay) where the new `2022.02.17.1` was unpullable through the tag `production`. The last deploy was "successful" in that it couldn't pull and rolled back to the last image it had access to.

During looking into this I noticed some integrations had incorrect Hub Names and malformed tokens.
This has been added: https://github.com/helium/console/issues/1061
And we're talking about what better error reporting for mqtt type connections.

DavidDeeds, 2022-02-26:
Yes! Confirmed working now - thanks so much!!!

closed 2022-02-26 by michaeldjeffrey