helium/router #636: Incrorrect downlink frame counter displayed in the console

Oliv4945, 2022-02-28:


macpie, 2022-03-08:
OK, I was able to reproduce this. Still working on tests to see what is going on. Thanks for the report.

[image]

mrx23dot, 2022-03-08:
Not sure how, but I might have broken the system :D
Negative Fcnt:  (should be unsigned based on standard)
[image]

device id: 66e67e40-90ff-4177-a23b-29c33f4d6c35

```
  {
    "category": "downlink",
    "data": {
      "devaddr": "08030048",
      "fcnt": -1,
      "hotspot": {
        "channel": 7,
        "frequency": 869.525,
        "id": "112L15XNSydYs9ccR5cyqfou3YtbTQSk89GLbmxpvykyyCgX6n1W",
        "name": "oblong-ultraviolet-starling",
        "rssi": 27,
        "snr": 0,
        "spreading": "SF12BW125"
      },
      "integration": {
        "id": "console_websocket",
        "name": "Console downlink tool",
        "status": "success"
      },
      "mac": [],
      "payload": "oAgDAEgAAAABmd/xbyz9MVWWygE=",
      "payload_size": 20,
      "port": 1
    },
    "description": "Confirmed data down sent",
    "device_id": "66e67e40-90ff-4177-a23b-29c33f4d6c35",
    "frame_down": -1,
    "frame_up": null,
    "id": "c14aae41-1bca-48ce-8f03-de7091d72880",
    "organization_id": "32595cfd-96a9-4510-96dd-826b546711db",
    "reported_at": "1646774266464",
    "router_uuid": "1bdeb46d-1546-4e58-8166-8ae878bd777c",
    "sub_category": "downlink_confirmed"
  },

```

macpie, 2022-03-09:
Issue should be fixed in https://github.com/helium/router/pull/649

[image]