helium/router #641: Poor uplink hit rate if if node is using BW500K for joining to Helium

mikedsp, 2022-03-04:
[](https://user-images.githubusercontent.com/47757569/156560916-2d78b15c-5a1d-4a62-94e7-5ed24ffe2fb1.png)
Last week I joined a Browan MerryIoT CO2 sensor to the Helium network and noticed that a significant portion of the uplink messages were not being picked up by the network. I have 2 hotspots in my house and several around me and the rest of my sensors typically have an uplink connection rate of 95+%. Upon further investigation I noticed that the datarate being used for the MerryIoT was SF8BW500. All my other 20+ sensors use BW125, so the BW500 was a surprise.

I reached out to the Browan application engineer and his response was as follows:
"I'm sorry to tell you that there is a known issue if node is using BW500K for joining to Helium. However, this issue comes from Helium LNS (LinkADR MAC command) instead of our nodes, it will cause the uplink hit rate issue. We've reminded and discussed with the Helium team, they are going to fix this issue and will keep us updated.
On the other hand, according to standard LoRaWAN US915 join procedure, the join request messages on random 125k and 500k, so in case of join successfully at 500k, it is probabilistic instead of our design.
The workaround is relaunching the OTAA join by holding the test button for over 5 seconds to see if it can just join the Helium on 125k."

I followed the recommended work around and the device joined with BW125 and since then the uplink hit rate has been fine.
I have attached a Log of messages received from the device. In the log you can see that when BW=500, there are are gaps in the frame count, but that once the device was rejoined with BW125, the gaps in the frame count go away.

**To Reproduce**
1. Join a MerryIoT CO2 sensor (US band) to the Helium network. 
2. Trigger an uplink by pressing the button on the device
3. If the datarate has BW125, the uplink will likely be picked up by the Helium network
4. If the datarate has BW500, there is a good chance the uplink will not be picked up by the Helium network. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Device Info (please complete the following information):**
 - Production VIP
 - Console Device ID: c5072f29-2a00-4272-838f-4aefd1a004b6
 - Device dev eui / app eui: 0016160000000218/0016160000000003
 - How often is device sending data: every 10 min
 - Region: US
 - Hotspots (names) used to transfer data: gorgeous-fuchsia-penguin, magic-latte-nuthatch, creamy-holographic-cat

*NOTE: do not post your device `app_key`*

**Additional context**
[20220303_Log_MerryIot_CO2.xlsx](https://github.com/helium/router/files/8182374/20220303_Log_MerryIot_CO2.xlsx)
[image]

Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

pvolodin, 2022-03-04:
Actually, it's the same as #633

jdgemm, 2022-03-04:
duplicate of #633

closed 2022-03-04 by jdgemm