helium/router #643: Fcnt rollover not correctly handled

mrx23dot, 2022-03-05:
Last valid packet up:
[image]

Since Fcnt is 16bit in frame, the next valid counter is 0, but it gets dropped.
Sending Fcnt=1 is dropped for the same reason.
Sending 65534, 65535 again is obviously dropped.

So my conclusion rollover is not handled on Helium.
It's easy to implement  for v1.0.4
`u16 diff = u16 new - u16 last // handles rollover`  
diff > 0

```
[
  {
    "category": "uplink_dropped",
    "data": {
      "fcnt": 65535,
      "hold_time": 623,
      "hotspot": {
        "id": "112K3K82t8VagR8kVmrPGqcQKftaxp2P9pMpaDKY7mDXwt5ToXpU",
        "name": "real-shadow-gecko"
      }
    },
    "description": "Late packet",
    "device_id": "66e67e40-90ff-4177-a23b-29c33f4d6c35",
    "frame_down": null,
    "frame_up": 65535,
    "id": "e4d7e1fb-f63a-4bc1-adfc-17974e8a7383",
    "organization_id": "32595cfd-96a9-4510-96dd-826b546711db",
    "reported_at": "1646480468954",
    "router_uuid": "829ae5ba-7690-47b5-ad4e-75fde0fd764d",
    "sub_category": "uplink_dropped_late"
  },
  
  -----------------------------------------
  {
    "category": "uplink",
    "data": {
      "integration": {
        "id": "no_channel",
        "name": "no_channel",
        "status": "no_channel"
      },
      "req": {}
    },
    "description": "No Channel Configured",
    "device_id": "66e67e40-90ff-4177-a23b-29c33f4d6c35",
    "frame_down": null,
    "frame_up": null,
    "id": "94ba9b82-6a53-44b3-b986-5c8935cc62e5",
    "organization_id": "32595cfd-96a9-4510-96dd-826b546711db",
    "reported_at": "1646480426775",
    "router_uuid": "2ec02a8b-ddfb-409a-8f8c-6f8dc7c1f8af",
    "sub_category": "uplink_integration_req"
  },
  {
    "category": "uplink",
    "data": {
      "dc": {
        "balance": 8711,
        "nonce": 1,
        "used": 1
      },
      "devaddr": "44000048",
      "fcnt": 65535,
      "hold_time": 1730,
      "hotspot": {
        "channel": 1,
        "frequency": 868.2999877929688,
        "id": "112waZ6nbEM6VhN6E1G1cpNm4VZRb98ZMi5qdA29U7qXeUaTkLb4",
        "name": "deep-sapphire-eagle",
        "rssi": -125,
        "snr": -24.5,
        "spreading": "SF12BW125"
      },
      "mac": [],
      "payload": "",
      "payload_size": 0,
      "port": 1,
      "raw_packet": "QEQAAEgA//8BvcfK4Q=="
    },
    "description": "Unconfirmed data up received",
    "device_id": "66e67e40-90ff-4177-a23b-29c33f4d6c35",
    "frame_down": null,
    "frame_up": 65535,
    "id": "1d075c30-575a-4d17-939d-d1b8d237a752",
    "organization_id": "32595cfd-96a9-4510-96dd-826b546711db",
    "reported_at": "1646480424380",
    "router_uuid": "2ec02a8b-ddfb-409a-8f8c-6f8dc7c1f8af",
    "sub_category": "uplink_unconfirmed"
  },
```
DevEUI 6081F987FC83BF62
App EUI 6081F90FDB670D39
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 - Console Device ID
 - Device dev eui / app eui
 - How often is device sending data
 - Region
 - Hotspots (names) used to transfer data

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

mrx23dot, 2022-03-05:
I assume same goes for downlink, but it would be expensive to test.

mikev, 2022-03-05:
fyi - our new lorawan library correctly handles this.

mrx23dot, 2022-03-06:
Are you saying it's a gateway bug and it won't be fixed until the owner updates the firmware?
Can you force firmware update? (since Helium requires approved gateways..)
What about redundancy, my uplink was received 5gateways all of them buggy?

Or is this checked on the server after the packer forwarder?

At this state every sensor will get stuck in 1.2years when reporting up every 10mins.

mrx23dot, 2022-03-06:
As I read:
`All the gateway does is to forward this encrypted+authenticated payload (which was sent over radio, over unencrypted UDP`
so it looks like it's a server thing, so why isn't easily fixed?

mrx23dot, 2022-03-07:
Also a better description than     "description": "Late packet", would be better, like 
"Frame counter invalid", 
"MIC invalid",

Oliv4945, 2022-05-04:
> fyi - our new lorawan library correctly handles this.

@mikev Any ETA on this? We also had a couple of sensors reaching a frame counter of 655535 that we now have to restart to get the back on the network

barndon, 2022-05-27:
Apologies for the delays on this.  Here is a brief update:

While I have not been able to reproduce the "Late packet" result described above, I _can_ show that Router's handling of frame counters (both FCntUp and FCntDown) is incorrectly limited to 16 bits.  I hope to have a fix for it in the upcoming Sprint 78.

closed 2022-06-24 by macpie