helium/router #644: [FR] Eliminate the need of IP port forwarding

mrx23dot, 2022-03-06:
This would make installation lot easier for end gateway user. (plug and play)
It wouldn't even work with GSM networks (NATed IPs).

Why is it needed? 
Gateway receives a packet, forwards it to server , checks for answers.
Server doesn't have to initiate.

Even a smart lamp bulb can solve this, you can control it over a .com domain outside your house without port forwarding.

closed 2022-03-08 by jdgemm

mrx23dot, 2022-03-10:
For the full hotspot you could use UPnP to try automatic port forwarding over router.
If it fails user can still doe it manually.

macpie, 2022-03-11:
> For the full hotspot you could use UPnP to try automatic port forwarding over router. If it fails user can still doe it manually. There is a linux lib for it.

This is already the case, see [here](https://github.com/helium/erlang-libp2p/tree/master/src/nat).