helium/router #645: Fport missing in downlinks

mrx23dot, 2022-03-06:
I queued a downlink with port 1:
[image]

It went down:
[image]

"c```
ategory": "downlink",
"id": "112nEViXNuPJTMkhBjoJoDusBH1zWV9SAPZTASRy6WquKSkzdYiZ",
"port": 1
"payload_size": 12,
"payload": "YFwAAEgAAgDTqSH2",
"payload_hex ": 60 5c 00 00 48 00 02 00 d3 a9 21 f6
```

But it doesn't have a port:
![2022-03-06_143559_X0](https://user-images.githubusercontent.com/4163396/156927830-ef54324e-90e6-4f69-a458-c79bcb2adfca.jpg)

pvolodin, 2022-03-06:
FPort missing in downlinks **if application payload length is 0**.

mikev, 2022-03-07:
The FRMPayload is empty, therefore per the spec, "A frame with a valid FHDR, no FOpts (FOptsLen=0), no FPort and no FRMPayload is a valid frame."

mrx23dot, 2022-03-07:
Thank you, we might want to visually indicate that in the manual downlink debug window.
(grey out port without user payload)

pvolodin, 2022-03-07:
@mikev 
> "A frame with a valid FHDR, no FOpts (FOptsLen=0), no FPort and no FRMPayload is a valid frame."

But this doesn't mean a frame with non-empty FPort and no FRMPayload is invalid. 

[image]

So it's a bug, @mrx23dot I believe you should reopen this.

mikev, 2022-03-07:
However the packet as described above is valid.  See the decode for confirmation:
https://lorawan-packet-decoder-0ta6puiniaut.runkit.sh/?data=YFwAAEgAAgDTqSH2&nwkskey=&appskey=

pvolodin, 2022-03-07:
> However the packet as described above is valid.

Surely it is valid. It just differs from the packet sent from application layer. And that's the issue.

pvolodin, 2022-03-07:
This is how correct packet should be looking like (though MIC is incorrect as I dont know the keys):
https://lorawan-packet-decoder-0ta6puiniaut.runkit.sh/?data=605C00004800020001D3A921F6&nwkskey=&appskey=

mrx23dot, 2022-03-08:
Just tested uplink accepts packet with port and without user-payload,
so why can we do the same in downlink?

mikev, 2022-03-08:
> Just tested uplink accepts packet with port and without user-payload,
> so why can't we generate the same in downlink?

The payload is empty and the spec allows either fport==1 or no fport.  No fport is more efficient from a library and transmit perspective, because you save one byte.  Is this a feature request?  Why do you want to generate such a Downlink packet?

mrx23dot, 2022-03-08:
I want to store information in Port, without additional user payload.

- Port 0 is reserved for MAC codes, but 
- Port 1 could say TURN_ON
- Port 2 could say TURN_OFF
- ..
- till Port ~244 which is reserved for testing again.

Based on lorawan standards this is a valid usecase.

pvolodin, 2022-03-09:
> The payload is empty and the spec allows either fport==1 or no fport.

Nothing in the spec allows the server to modify application-specific data. And FPort values 1..223 are application-specific.

> * till Port ~244 which is reserved for testing again.

224, to be exact

Oliv4945, 2022-03-11:
Hi @mikev I also agree that the NS should not remove the port if the payload size is 0

closed 2022-09-07 by jdgemm