helium/router #651: Queue Downlinks Before Join

kent-williams, 2022-03-10:
A common strategy for device deployments is to queue downlinks prior to customers/users receiving the device and it joining for the first time. This is possible on other LNS, but not yet with Console/Router. This places an extra burden on the device manager to create logic for queuing downlinks after a join. Is this something we could allow without much effort?

macpie, 2022-04-15:
@kent-williams The problem here is until we get a join we do not know the region of the device so we cannot properly calculate `Data Rate Down`. We would need to provide the region to downlink then and change UI, it that something you feel we should do?

macpie, 2022-04-15:
@kent-williams I was able to queue a downlink before a join could you confirm it is working on your end?

kent-williams, 2022-04-25:
@macpie Are you able to queue before the very first join? I believe @michaeldjeffrey clarified this in chat.

macpie, 2022-04-28:
We have decided to add an option to include region when sending a downlink. Router and Console will need to support this.

macpie, 2022-05-13:
Fixed in https://github.com/helium/router/pull/732

closed 2022-05-13 by macpie