helium/router #652: Unable to use max downlink MTU

Oliv4945, 2022-03-11:
**Describe the bug**
The maximum downlink MTU can not be used

**To Reproduce**
1. Set a fix datarate. ex; DR0 in EU868 region
2. Create a downlink using all the MTU. ex: 51 bytes

**Expected behavior**
Downlink is sent

**Current behavior**
The downlink is not sent, error message is 
```
  "category": "downlink_dropped",
  "description": "Payload too big for DR2 max size is 51 (payload was 51)"
```

**Additional context**
I am not skilled in Erlang but I believe that the comparison on [this line](https://github.com/helium/router/blob/e5fdd9ee7471ed7fa934594256ff37a8c124e0cc/src/device/router_device.erl#L440) should be `<=` instead of `<` so the max MTU is allowed:
```
            Region = ?MODULE:region(Device),
            UpDR = ?MODULE:last_known_datarate(Device),
            DownDR = lorawan_mac_region:dr_to_down(Region, UpDR, 0),
            MaxSize = lorawan_mac_region:max_payload_size(Region, DownDR),
            Size = erlang:byte_size(Payload),
            {Size < MaxSize, Size, MaxSize, DownDR}
```

macpie, 2022-03-17:
> **Describe the bug** The maximum downlink MTU can not be used
> 
> **To Reproduce**
> 
>     1. Set a fix datarate. ex; DR0 in EU868 region
> 
>     2. Create a downlink using all the MTU. ex: 51 bytes
> 
> 
> **Expected behavior** Downlink is sent
> 
> **Current behavior** The downlink is not sent, error message is
> 
> ```
>   "category": "downlink_dropped",
>   "description": "Payload too big for DR0 max size is 51 (payload was 51)"
> ```
> 
> **Additional context** I am not skilled in Erlang but I believe that the comparison on [this line](https://github.com/helium/router/blob/e5fdd9ee7471ed7fa934594256ff37a8c124e0cc/src/device/router_device.erl#L440) should be `<=` instead of `<` so the max MTU is allowed:
> 
> ```
>             Region = ?MODULE:region(Device),
>             UpDR = ?MODULE:last_known_datarate(Device),
>             DownDR = lorawan_mac_region:dr_to_down(Region, UpDR, 0),
>             MaxSize = lorawan_mac_region:max_payload_size(Region, DownDR),
>             Size = erlang:byte_size(Payload),
>             {Size < MaxSize, Size, MaxSize, DownDR}
> ```

Good catch, fixing.

macpie, 2022-03-17:
@Oliv4945  I know Erlang can be scary but please feel free to make pull request on issues like this. We would be more than happy to fast track contributions!

closed 2022-03-17 by macpie

Oliv4945, 2022-03-18:
> @Oliv4945 I know Erlang can be scary but please feel free to make pull request on issues like this. We would be more than happy to fast track contributions!

Ok thanks. I thought about it but I might be unable to create an associated test...
I'll consider it next time :)

PR #654 (2022-03-17): Allow max downlink MTU
Fix #652