helium/router #664: HIP45 - Australia and Malaysia

mikev, 2022-03-23:
Parent ticket for work related to HIP45 including Australia, Malaysia and Region plan customization.

Australian-Hnt-collectors, 2022-03-25:
If we are discussing this HIP here as well should we not join the https://github.com/dewi-alliance/hplans/pull/28 discussion as this is the Australian peoples views on any potential changes just a question.

ehrisbok, 2022-03-25:
Why is Australia being now put in with Malaysia?? This topic is already being discussed regarding Australia & New Zealand as noted by the previous post.

mikev, 2022-04-18:
Related work
https://github.com/helium/router/commit/ffe3dc28e38fd8a09c6a1bf7bb307789311f7994

oberchoo, 2022-04-23:
Hi, am from Malaysia. Currently, Malaysia only allows operating freq from 919 up to 924MHz. However, LoRa Alliance assigned AS923-1 for Malaysia. And if we go detailed, AS923-1 is: 923.2, 923.4, 923.6, 923.8, 924.0, 924.2, 924.4, 924.6 MHz. So I believe the plan here is to develop a script or program to restrict hotspots in Malaysia to operate within 923-924MHz, which is still in compliance with AS923-1 as assigned by LoRa Alliance and legal in Malaysia. In other words, once a hotspot is detected in Malaysia (via the location asserted during registration of the hotspot), it will be limited to operating from 923.2 - 924MHz.

https://github.com/helium/HIP/issues/311 

@mikev, I am looking forward this to being implemented. Any estimation of deployment date? If you need any help or test, I am more than happy to do it.

mikev, 2022-04-25:
This work will include support for IN865

karthiclodpi, 2022-04-26:
> This work will include support for IN865

Frequency plan :

Uplink:
865.0625 - SF7BW125 to SF12BW125
865.4025 - SF7BW125 to SF12BW125
865.9850 - SF7BW125 to SF12BW125

Downlink:
Uplink channels 1-3 (RX1)
866.550 - SF10BW125 (RX2)

@mikev

mikev, 2022-04-29:
This is the parent ticket where external parties can track progress.

karthiclodpi, 2022-05-03:
Hi @mikev 

IN865 Devices are not still joining. Join Request is not reaching Console

mikev, 2022-05-05:
@karthiclodpi 
As I stated earlier, IN868 is not supported currently.  There will be an announcement when IN868 is supported.  ETA is likely August 2022 or later.

mikev, 2022-05-05:
**Progress update**
We completed integration with the new LoRaWAN library, specifically the lora_plan module which now contains frequency plan logic.  The new logic is data-driven so that new frequency plans can be devised without changing the code.
https://github.com/helium/router/pull/710

mikev, 2022-05-10:
New AS923_1B configuration updated in sx1302 project.  This firmware update configures the Miner's listening channels.
https://github.com/helium/sx1302_hal/commit/dcbeb20711579ba75a3ae8ce6624725cf8064da6

mikev, 2022-05-16:
Region enums are now updated in Proto and Gateway-RS
https://github.com/helium/gateway-rs/pull/250
https://github.com/helium/proto/pull/132

mikev, 2022-05-18:
Miner update
https://github.com/helium/miner/pull/1664

mikev, 2022-05-18:
Blockchain update
https://github.com/helium/blockchain-core/pull/1355

mikev, 2022-05-18:
H3 hex region update
https://github.com/helium/lorawan-h3/commit/1875b682aabf45b8679f083a67cb94e85a52d227

mikev, 2022-05-22:
LoRaWAN library update merged in Router master
https://github.com/helium/erlang-lorawan/pull/6

mikev, 2022-06-20:
Router logic
https://github.com/helium/router/pull/790

closed 2022-09-07 by jdgemm

mcauser, 2022-11-15:
- [ ] Hardware QA testing

Not needed?

leogaggl, 2022-11-15:
> * [ ]  Hardware QA testing
> 
> Not needed?

Of course not. Not at Helium. Testing is something for mere mortals.

> At the risk of oversimplifying, Helium develops systems based on mathematical models where we know basic math and model facts, e.g. 2 + 3 == 5.  We don't need to physically test to understand and know that our underlying model will actually work in the real world.

And thay also don't actually need to communicate this to the affected community. 

Software engineering and change management of the finest!

abhay, 2022-11-15:
@leogaggl it appears you've decided to come here and troll. That's a fine choice. Have you considered adding the rest of the snippet in that email sent to you and cc-ed a few other folks? I asked around and I got the email so here it is for reference since you decided to cherry pick:

> Hi Leo,
>
>> there is no information on how this will work in the real world. One device in the lab might be encouraging, but there are no details to judge this with real-world devices
> 
> At the risk of oversimplifying, Helium develops systems based on mathematical models where we know basic math and model facts, e.g. 2 + 3 == 5.  We don't need to physically test to understand and know that our underlying model will actually work in the real world.
> 
> India, China, Korea, Malaysia and their associated plans are operational. were all developed using this approach.  The India plan IN865 was recently deployed without development using a physical device.  Testing was performed within India by volunteers.   It was the same with Malaysia.  There is a precedent for having locals in the region do final testing.
AU915 is almost identical to US915 but with some odd differences.  For development we use a virtualized environment which allows us to test different plans without having actual physical devices.
> 
> However in the case of DualPlan I did perform physical testing using a real AU915 radio device and a real DualPlan Hotspot.  All testing I performed used real hardware devices and Gateways.

So there was testing. Sure @mikev should have probably not given you the perfect snipped to quote and shout from the rooftops but your trolling here is unproductive.

Here's the rest of the email as well for reference:

>>  Is there some testing and user acceptance plan?
>
> Yes, we are in the DualPlan Beta phase now.  This is the reason for reaching out to yourself and others on the email thread.  We'd like select members of the Australian community to try out the system, give us feedback and judge for yourself if it works or not.  Further we have other Australian customers who are also currently Beta testing.

So it looks like this is an actual attempt at solving the problem you are pointing out:

> And they also don't actually need to communicate this to the affected community.

It sounds like you have strong opinions about an SDLC which is great! That said, you appear to be spending more time trolling than being helpful. It must be frustrating.

mikev, 2022-11-15:
DualPlan has been extensively QA tested.  It was tested internally by two different QA engineers.  It went through a Beta test period and multiple hardware Makers volunteered to Beta test and all reported that it worked perfectly.  Plus multiple community members also Beta tested.

The one known issue is SB2 vs SB6 which we've been transparent about in our announcements.

michaeldjeffrey, 2022-11-17:
> * [ ]  Hardware QA testing
> 
> Not needed?

Thanks for pointing this out. Sometimes tracking things like this get left behind as we reorganize tasks as they evolve.