helium/router #669: MAC commands radio data should be transmitted to the AS

Oliv4945, 2022-03-25:
**Describe the bug**
No message is sent to the AS when MAC command are sent on port 0. This is an issue for example for the AS which can not compute accurate PER statistics as those packets are correctly received by the NS, but the AS is not currently aware of it.

**To Reproduce**
1. Setup an HTTP integration
2. Setup a device & join the network
3. Send regular payloads (FPort <>0)
4. Set a payload with MAC commands only (FPort=0)

**Expected behavior**
I expect the AS to receive the same message on step 4 than step 3 expect the payload itself, for example:
```
{
   "app_eui":"0102030405060708",
   "dc":{
      "dc_balance":808978,
      "nonce":2
   },
   "dev_eui":"0102030405060708",
   "devaddr":"9E000048",
   "fcnt":29989,
   "hotspots":[
      "<REDACTED>"
   ],
   "id":"95134a0f-a6aa-4c86-a007-86a796688c36",
   "metadata":{
      "adr_allowed":false,
      "cf_list_enabled":false,
      "labels":[
         "<REDACTED>"
      ],
      "multi_buy":1,
      "organization_id":"b298aa5c-b65f-4585-9dff-c55792f31d97",
      "rx_delay":1,
      "rx_delay_state":"rx_delay_change"
   },
   "name":"Elsys Desk 1",
   "payload":"Bw4dEQA=",
   "payload_size":5,
   "port":5,
   "raw_packet":"QJ4AAEiAJXUFa9v1BuiheHev",
   "replay":false,
   "reported_at":1648221036260,
   "type":"uplink",
   "uuid":"b7ae1d3b-89ae-4228-8ed5-8a25942639ce"
}
```

**Screenshots**

[image]

macpie, 2022-05-23:
Port 0 is disabled by default on most server. I need to understand usage better before allowing usage of Port 0.

Note: Router is an Application Server (AS) and Network Server (NS) at the same time

jamesl-dm, 2022-05-24:
One possible usage is rough geolocation using the receiving gateway location. Another niche use case is determining that no information was lost in a run of packets. For instance, if a fragmentation protocol requires that packets be delivered consecutively with no gaps in order to guarantee reassembly, the gap created by a port 0 uplink not being reported would be a problem. Some devices may also send an empty packet (with no port number) occasionally, just to signal that they're still alive. If a MAC command happens to be present in the uplink, it might be converted to a port 0 uplink, and not delivered to the application server.

Might make sense to treat the uplink as an empty packet with no port number, if you'd rather not report a port number of zero with no payload.

Oliv4945, 2022-05-24:
Hi, James nicely resumed the use cases. The third one I would add is packet error rate figures, currently it can not be correctly computed as packets with Fcnt=0 are not provide to the application.

> Port 0 is disabled by default on most server.

I am not sure that the payload containing the MAC command should be provided to the user, only the metadata. For example this is how TheThingsStack is doing.

Oliv4945, 2022-06-24:
hey @macpie  @michaeldjeffrey thanks, but your merge request does not answers to the issue: the issue is about uplinks from the device that are not provided to the AS.

Could you please reopen it?

closed 2022-09-07 by jdgemm