helium/router #677: Benchmarking

dpezely, 2022-04-01:
Load & Capacity Measurement 
(some people call this "stress testing")

- Use multiple instances of [virtual-lorawan-device](https://github.com/helium/virtual-lorawan-device)
  + e.g., [Debug with sniffer](https://docs.helium.com/use-the-network/run-a-network-server/debug-with-sniffer) tutorial
- Use multiple Organizations, each with a unique `NetID`
- Intentionally induce `DevAddr` collisions across these different (fake) organizations
- Measure load on an isolated instance of Router to determine practical thresholds for LoRaWAN traffic capacity
- Attempt to simulate real-world use cases

Tasks:
- [ ] Implement on dev router instance
- [ ] Measure on various small-ish AWS server instances

lthiery, 2022-04-02:
It's of interest, I have an OUI with 8 devaddr slab running on Hetzner. I can provide access

dpezely, 2022-04-18:
A minor issue: router seems to only permit one gateway coming from a single IP address.

The solution could be simply to run gateway-rs from multiple AWS server instances (e.g., spot instances) which would also avoid resource contention from a single host sending all those uplinks.  Multiple hosts was initially avoided to reduce complexity.

dpezely, 2022-04-19:
Finally have all configured devices utilizing all gateway-rs instances on a single machine.  (There was an off-by-one error in a Bash excerpt used for generating IP port number offsets.)

We also have a few AWS server instances ready for running there, which will reduce Internet latency.

dpezely, 2022-04-20:
On the larger run with hundreds of devices from an AWS server instance running gateway-rs and virtual-lorawan-device, the vast majority of devices were unable to Join.

Investigating...

dpezely, 2022-04-21:
There seems to be a resource limit on number of `gateway-rs` inbound socket connections (and therefore limits number of concurrent `virtual-lorawan-device` instances per gateway instance).

Investigating gateway-rs code and dependencies (e.g., Tokio), but Linux kernel vars seem sufficient (i.e., `somaxconn`).

madninja, 2022-04-21:
huh? why would a single gateway-rs need to handle more than a few connections?

dpezely, 2022-04-21:
> huh? why would a single gateway-rs need to handle more than a few connections?

This is for driving artificial load for sake of determining capacity of a particular size AWS server instance running `router`. Yes, it's far beyond the originally intended use cases for `gateway-rs`, so 100 devices per organization means hundreds of concurrent connections.

madninja, 2022-04-21:
> This is for driving artificial load for sake of determining capacity of a particular size AWS server instance running `router`. Yes, it's far beyond the originally intended use cases for [...]

uhh.. isn't that just the virtual lorawan device simulating hundreds of devices but it's still a single connection to gateway-rs for that packet forwarder? What am I missing?

dpezely, 2022-04-21:
> virtual lorawan device simulating hundreds of devices

Yes, this looks like the best path.  Either way, there will be new Rust code!
(I was first exploring the least coding effort, but that's been sufficiently exhausted now.)

dpezely, 2022-04-22:
Now that we're using `virtual-lorawan-device` for driving 100 devices per organization, driving load at scale is successful.

This also confirms abuse-prevention measures of the new reputation system, and that gets triggered within a minute or so. (Reputation score exceeded 50 in about 1 minute).

Packet uplink offers are getting dropped by router due to "devaddr not in subnet", so @lthiery it may be time to take you up on that kind offer of using your 8 devaddr slab.

dpezely, 2022-04-26:
Clarified my understanding of a few points and updated the docs PR with more to come.

e.g., Location distances only comes into play when same device sends via different gateway, but we do not yet have the means to have round-robin *per device* uplinks.  I'll put that feature on my to-do list.

Adding more instances of gateway-rs and asserting their locations to an obviously bogus location such as a six meters above a lake.

dpezely, 2022-04-26:
On a non-trivial but light run, Grafana indicated 98%-99% for **Packet Offer Rate** success.
This run involved **5 gateways, 2 organizations each with 100 devices** all driven from same region within AWS to minimize Internet latency.

The graph dipped briefly to 95%-96% a few times within the first ten minutes and then down towards 75% for next ten minutes.

Meanwhile, even though router's reputation response was temporarily disabled yet still keeping score, these 5 gateways each have a score in the thousands; e.g., 4k, 5k after only ten minutes; over 10k around 15 minute mark;

Run began at time = Tue Apr 26 22:04:01 2022 GMT and ended at 22:28:26  GMT.

This is NOT using a purely dedicated server for benchmarking yet.

dpezely, 2022-04-29:
An issue with benchmark runs is that occasionally a gateway-rs instance or two will disconnect from its upstream validator.  When this happens, traffic drops-off because devices going through that gw can't reach router.

That apparently happened with yesterday's run and again just now.

We may have to wait to get real results until after gateway-rs v1.0 is official (still v1.0 alpha at the moment, according to its repo).

There is an upcoming release for Validators, gateway-rs, etc.

My understanding is that newer Validator code will listen on changed/additional IP ports that are not yet open.  This means that upon starting gateway-rs today, it will sometimes take an hour to connect because of trying (currently) invalid port numbers.

dpezely, 2022-05-03:
Regarding **devaddr allocation with an 8 slab**:
- 200 devices total was fine, third set of 100 were unable to Join
   + 2 separate instances of gateway-rs
   + each gateway-rs on a unique Public IP address
   + 100 devices per Organization
   + each org unique to a single instance of virtual-lorawan-device
   + Each device: `default_secs_between_transmits` = 15
- Third org containing 100 saw 100% failure to join
   + via new instance of virtual-lorawan-device
   + hitting an existing gateway-rs already servicing first org with 100 devices
- After adding third org, traffic for first two orgs were apparently unharmed
- More fine-grained tests have yet to be run for finding more precise value
   + smaller tranche sizes; e.g., 100 -> 50, 25, 15, 10, 5 

When using lower time between transmits such as 10 or 5, there were too many late downlinks, so either gateway-rs or virtual-lorawan-device would need to be patched.

Note: 

Neither gateway-rs nor virtual-lorawan-device were intended to be used for this use-case, so we may ultimately need to modify one or the other.  While some transient errors are expected, I'm exploring a feature of virtual-lorawan-device to steadily increase pre-configured devices (rather than current behavior of all-or-nothing) while remaining within a configurable error rate, defaulting to 1:500 or so.

dpezely, 2022-05-04:
The upper limit seems to be **approximately 2250 devices** for an 8 slab devaddr, but *results still need to be replicated* and bracketed.

To sanity-check my thought process here: despite some noise upon starting each batch, once each batch's Joins have all settled, I take that to mean the devaddr 8 slab + MIC disambiguation resolution has absorbed the pool of new devices.

The last 100 devices-- of which only about half were able to Join-- took dozens of attempts to Join.  This is likely due to our own traffic interfering with itself while abusing gateway-rs and the artificial nature of this simulation.

This latest run used the following configuration:
- 3 hotspots on AWS instance 1 
- 2 hotspots on AWS instance 2
- 1k devices in first org on instance 1 spraying across those 3 hotspots
- 1k devices in first org on instance 2 spraying across those 2 hotspots
- added another org, each with 100 devices per AWS host, also spraying across hotspots
- Upon adding that final 100 devices, it's estimated that half were never able to Join

mfalkvidd, 2022-05-05:
Very ineresting work, well done!

Did you record cpu usage data for the router? I'm curious whether the bottleneck is cpu power (so larger reuse would be possible by adding more cpu cores, or faster cores if the bottleneck is single threaded), or if the bottleneck is caused by collissions (more than 1 device key matches the mic, resulting in ambiguity that cannot be resolved).

dpezely, 2022-05-11:
Status update and highlights of **corrections**:

- Prior benchmark runs had non-conforming App EUI and App Keys for devices (must be unique)-- now fixed
- Prior runs intending to have more than zero seconds between device transmits remained at zero-- now fixed
- Router's method of tracking unique devices has been revised-- let's just say it's been battle-tested...
- Since previous update, use of a multiplexer has been added:
    + More accurately reflects real-world use cases because each device's packets go through *all* routers known to the muxer
    + Prior config generation remains in doc but commented-out as optional
- Draft doc has been revised (also linked from ticket description)
    + https://deploy-preview-846--helium-docs.netlify.app/use-the-network/run-a-network-server/router-benchmarking/

dpezely, 2022-05-13:
Update after several successful runs:

- First, `router-dev` does **not** have an 8 slab devaddr block but instead is a **1024 slab**
  + @mfalkvidd , I misspoke previously
  + This is why the count was so much higher than you expected
- The upper limit on concurrent devaddr allocations on `router-dev` seems to be **5800**
  + We still need to reproduce for confirmation of the number
  + The actual count was 5806
  + We highly encourage the Community to attempt reproducing our results and reporting
- Interpretation of graphs in Grafana from which we reached that number has been updated in the doc
- Final updates to the doc include a few lines of Bash for leveling traffic load on Router and host sending traffic
- At time of writing this post, there is an issue with upstream Validators that currently impacts further testing
  + Similarly, text and procedures have been added to the doc for diagnosing such issues
  + e.g., using API into `gateway-rs` instead of extracting from its logs for more reliable results
- Thank you, everyone, for your patience with this endeavor!
    + Planning and executing this is *unlike* its equivalent for HTTP traffic
    + The number of experimental conditions and variables to track with each run is like doing multi-variate analysis or holding a large polynomial equation in one's head, but hopefully the doc accommodates copy/paste

closed 2022-09-07 by jdgemm