helium/router #680: Grafana dashboard template does not match prometheus database

disk91, 2022-04-03:
The grafana dashboard refer to server but the prometheus database refer the data with instance.
The prometheus.yml file may not be configured correctly to match the grafana dashboard.
You can find a modified version of the grafana dashboard that work with the prometheus setting here : https://www.disk91.com/wp-content/uploads/2022/04/grafana_helium_router.json_.zip

michaeldjeffrey, 2022-04-03:
The instance label is something Prometheus adds based on the endpoint the data was scraped from. The server field is something added globally from the Prometheus config. 

We use that to differentiate data from different servers because they all pull from the same endpoint and show up and the same instance.

disk91, 2022-04-04:
Great, so the Prometheus config should be adapted for this

michaeldjeffrey, 2022-04-04:
Maybe I'm not understanding. 
That's what [`promethues-template.yml:L5`](https://github.com/helium/router/blob/master/prometheus-template.yml#L5) does.

macpie, 2022-04-11:
> Maybe I'm not understanding.
> That's what [`promethues-template.yml:L5`](https://github.com/helium/router/blob/master/prometheus-template.yml#L5) does.

@disk91  Did this resolve the issue?

closed 2022-05-23 by jdgemm

macpie, 2022-05-23:
FYI: Template will get updated as new metrics are added