helium/router #682: router organization info all does not display the org UUID correctly

disk91, 2022-04-05:
When you use command

```
docker exec helium_router router organization info all
|fc40xxx-xxxxx-xxxxx-xxxxx-aaf31b|       xxxxxxx        |undefin| 0  | undefined  |undefined |
```
you get an oragnization number but when you do:

```
docker exec helium_router router organization info fc40xxx-xxxxx-xxxxx-xxxxx-aaf31b
Organization not found.
ok
```

So there is a problem somewhere

macpie, 2022-04-26:
@disk91 Is this still an issue? I was unable to reproduce.

disk91, 2022-04-27:
Yes : the reason is info all does not display the organization id full but as a substring
ex : 
```
what is returned
|ad00a606-xxxx-xxxx-b312-adf433|          Helium IoT      
the real id
ad00a606-xxxx-xxxx-b312-adf433xxxxfa
```

michaeldjeffrey, 2022-04-27:
The table printer will do its best to fit the results in the space you give it. 
Does this continue to happen if you make your terminal wider?

closed 2022-04-28 by macpie