helium/router #683: When created a new org, the DC cached_balanced sometime is null, device can't communicate

disk91, 2022-04-05:
```
docker exec helium_router router organization info 7bd91c73-63cxxxxxxxxx
+------------------------------------+--------------+-------+-----+--------------+------------+
|                 id                 |     name     |balance|nonce|cached_balance|cached_nonce|
+------------------------------------+--------------+-------+-----+--------------+------------+
|7bd91cxxxxxx | xxxx |1000001|  1  |     null     |     0      |
+------------------------------------+--------------+-------+-----+--------------+------------+
```

Not sure if it is easy to reproduce, it seems to happen when you create an account, then add an organization, this org have 0DC and when you add DCs on it you get null.
This crash the routeur when getting traffic and the communication are lost

michaeldjeffrey, 2022-04-06:
> This crash the routeur when getting traffic and the communication are lost

Do you have, or can you get a crash log for this?

michaeldjeffrey, 2022-04-26:
Closing for merged PR and lack of activity.

closed 2022-04-26 by michaeldjeffrey