helium/router #687: Including CFList in LinkAdrReq when it's already in the JoinAccept

lthiery, 2022-04-10:
When I have a device that has CFList via JoinAccept enabled (US915), I can see that Router still proceeds with the routine of trying to send LinkAdrReq in response to the Data Frames.

This is redundant as the device is assumed to have received and processed the JoinAccept if it is sending a Data Frame.

macpie, 2022-04-11:
Will fix

pvolodin, 2022-04-18:
> I can see that Router still proceeds with the routine of trying to send LinkAdrReq in response to the Data Frames.

Are you sure there's no need to change the device's DR/TX power? In this case ChMask shall also be present.

Oliv4945, 2022-04-22:
Be careful, as there is no way to provide the device's LoRaWAN specification version to the router, this will be a breaking change:

I have devices using the same profile (with ADR on) that are v1.0.2, v1.0.3 and v1.0.4. The v1.0.3/v1.0.4 parse the CFList and use the correct channel plan, but the v1.0.2 will ignore it (as per specification), for those the subsequent LinkAdrReq is useful.

1.0.2 regional parameters:
[image]

lthiery, 2022-04-22:
1.0.2 devices should have a LoRaWAN profile on Console that has CFList disabled in that case? 

I guess it's not too bad of a thing to maintain for backwards compatibility, but if we are doing so, perhaps we remove the CFList toggle altogether.

closed 2022-09-07 by jdgemm