helium/router #688: ADR appears not to work for some sensors

mikedsp, 2022-04-11:
**Describe the bug**
ADR appears not to be working for 3 sensors on the Production network and 1 sensor on the VIP network. Three of the sensors are Browan TBWL00 water leak detection sensors and the other is a Browan Indoor Air Quality sensor of type TBHV110. 

From past experience, I've learned that for the Browan sensors to maintain high packet completion rates, ADR must be enabled. As a result, all my Browan sensors run with ADR enabled. If ADR is not enabled, the Spreading Factor gradually increments to SF10, at which point the packet completion rate drops from near 100% to approx 20%. With ADR enabled, shortly after SF increments to 10, the SF adjusts to a lower SF and the packet completion rate stays near 100%.

For the 4 Browan sensors mentioned above, SF has been stuck at 10 for multiple days and the packet completion rate has been poor the entire time. 

**To Reproduce**
Its hard to reproduce. Some Browan sensors on my console accounts are maintaining high packet completion rates with the SF auto-adjusting as needed while these 4 sensors do not. 

**Expected behavior**
The expected behavior for each of these sensors is that a short time after SF changes to 10 and packets start to get lost, SF drops to a lower number. 

**Device Info (please complete the following information):**
- Environment: Production (Org = mikedsp)
 - Console Device ID: TBWL100_8
 - Device dev eui / app eui: E8E1E1000104DE8E / E8E1E10001013641
 - How often is device sending data: once an hour
 - Region: US
 - Hotspots (names) used to transfer data: restless-champagne-orca

- Environment: Production  (Org = mikedsp)
 - Console Device ID: TBWL100_9
 - Device dev eui / app eui: E8E1E1000104DE81/ E8E1E10001013641
 - How often is device sending data: once an hour
 - Region: US
 - Hotspots (names) used to transfer data: restless-champagne-orca

- Environment: Production  (Org = mikedsp)
 - Console Device ID: TBWL100_10
 - Device dev eui / app eui: E8E1E1000104DEFB/ E8E1E10001013641
 - How often is device sending data: once an hour
 - Region: US
 - Hotspots (names) used to transfer data: restless-champagne-orca

- Environment: VIP (Org = Hitechdb)
 - Console Device ID: TBHV110_7
 - Device dev eui / app eui: 58A0CB000011E398/ 58A0CB0000210000
 - How often is device sending data: every 5 minutes
 - Region: US
 - Hotspots (names) used to transfer data: zesty-chili-raccoon, nutty-topaz-hamster

*NOTE: do not post your device `app_key`*

**Additional context**
You can use my Longview Dashboard to see the issue for the TBWL100 sensors. Use the Consolidated System view tab to see packet completion rates for all sensors in the system. It will be obvious that the 3 TBWL sensors listed above are performing poorer than the other Browan sensors in the system. Then use the room-specific dashboard tabs to drill into performance of each sensor. Look at the SF and Fcnt graphs. You'll see that when SF=10, there are longer flat lines in the Fcnt graphs indicating lost packets. You can compare/contrast the 3 poor performing sensors with the SF and Fcnt graphs of the other sensors in the system whose SF is not equal to 10.
https://app.datacake.de/dashboard/d/2ff8ac3e-8bd1-4f63-8f57-c351d2852f58

Similarly, you can use my LSSU dashboard to graphically analyze the performance of the TBHV110_7 sensor. 
https://app.datacake.de/dashboard/d/714777e6-5547-4e02-b0e5-563e4096c60b

If helpful, I can provide historical log files that contain network related data (time, hotspot, rssi, snr, SF) for every message received by the Helium network for these sensors. 

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.
[image]
This image is for sensor TBHV110_7 from the Library tab of the LSSU dashboard. You can see that for the past 24 hours, SF=10 and that rather than seeing a consistent staircase pattern on the fcnt graph, some of the steps are longer, indicating missed messages.

macpie, 2022-04-11:
From experience, I know that Browan sensors can be finicky with ADR. Can I ask what sensor you are using? Is it the GPS?

mikedsp, 2022-04-12:
Three of the sensors are Browan TBWL00 water leak detection sensors and the other is a Browan Indoor Air Quality sensor of type TBHV110.

mikedsp, 2022-04-29:
> Any idea why the Browan sensors loose packets at SF10? Usually a network will respond to packet loss by decreasing the datarate, not increasing it, so it makes sense to me that they should stay at SF10 if they're not performing well. So it sounds like a device issue to me.
>
> If there's no way to configure the Browan sensors so stay at SF7 and not use ADR, you'll probably need to configure the ADR's target spreading factor manually on the network side to make it reliable. You can now do this on TTN, but I don't think it's possible on Helium yet. Otherwise, even if it starts up on SF7, the first time you have a network outage it will drop back down to SF10 and stay there.

I don't know why Browan sensors lose packets at SF10. Before I enabled ADR, all my Browan sensors slowly drifted up to SF10 over a matter of days, then once they hit SF10 they stayed at SF10 for months with a 10-20% packet completion rate. Once I enabled ADR, over a period of days to weeks, the Browan sensors all eventually self-adjusted their SF off SF10 and packet completion returned to 90%+. After that, the SF would occasionally hit SF10 again, but would self-adjust off SF10 within a couple of hours. The reason I reported this issue is because for a couple of my Browan sensors the self-adjustment stopped and the sensor was stuck at SF10. Sensor TBWL100_8, for example, was stuck at SF10 for 10 days from April 4 - April 24.

macpie, 2022-05-09:
This seems more like a Brown issue will close for now

closed 2022-05-09 by macpie