helium/router #689: Network Time Request reply

EwanBez, 2022-04-11:
Don't see a way to allow for network time request reply to be sent on the next down link.
Expected to find it in profiles maybe - similar to ADR or RX slot time delay? 
[image]

macpie, 2022-04-11:
I do not think we support the `network time request` at this moment.

Oliv4945, 2022-04-13:
Hi, I also need this for some projects.

This MAC command has been added in LoRaWAN spec v1.0.3, what is your current plan regarding 1.0.4 @macpie? I saw #445 but it seems to be stalled

closed 2022-09-07 by jdgemm

Sensative-Lars-Mats, 2022-10-09:
Solving https://github.com/helium/router/issues/637 does not solve this issue in itself. The need is really to support 1.0.3 or 1.0.4 LoRaWan.