helium/router #701: Router can't download the snapshot

CanEmreUnal, 2022-04-19:
Router is not able to get snapshot from helium. Getting following error.

We just want to run self hosted instance but router is not syncing itself to blockchain.

```
[ec2-user@ip-*********** ~]$ docker exec helium_router router info height
133		9200
```

Running latest tag.

`[error] [blockchain_worker:fetch_and_parse_latest_snapshot:{1125,21}] Couldn't fetch latest-snap.json because {error,nxdomain}: [{blockchain_worker,get_latest_snap_data,2,[{file,"/opt/router/_build/default/lib/blockchain/src/blockchain_worker.erl"},{line,1186}]},{blockchain_worker,fetch_and_parse_latest_snapshot,1,[{file,"/opt/router/_build/default/lib/blockchain/src/blockchain_worker.erl"},{line,1122}]},{blockchain_worker,start_snapshot_sync,1,[{file,"/opt/router/_build/default/lib/blockchain/src/blockchain_worker.erl"},{line,1065}]},{blockchain_worker,snapshot_sync,1,[{file,"/opt/router/_build/default/lib/blockchain/src/blockchain_worker.erl"},{line,868}]},{blockchain_worker,handle_info,2,[{file,"/opt/router/_build/default/lib/blockchain/src/blockchain_worker.erl"},{line,640}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,695}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,771}]},{proc_lib,wake_up,3,[{file,"proc_lib.erl"},{line,236}]}]`

macpie, 2022-04-19:
This looks like a DNS issue, Make sure you either have:

- `network_mode: host` in your `docker-compose.yaml`
```
  router:
    image: quay.io/team-helium/router:latest
    container_name: helium_router
    restart: always
    env_file: .env
    network_mode: host
```
-  Or if NOT using `network_mode: host` set  `dns: X.X.X.X` in your `docker-compose.yaml`
```
  router:
    image: quay.io/team-helium/router:latest
    container_name: helium_router
    restart: always
    env_file: .env
    dns: X.X.X.X
```

CanEmreUnal, 2022-04-19:
thanks it solved!

closed 2022-04-19 by CanEmreUnal