helium/router #704: Http Integration Not Working

CanEmreUnal, 2022-04-20:
After device has joined the network, uplink message is not forwarding to the endpoint for following reason.

Dig query finding the record on host server. Using 8.8.8.8

`[2022-04-20 21:53:26.602] <0.3371.0> [info] [229f4392-7ca4-4f16-a622-170ca038b7b0] [router_http_channel:check_url:{197,29}] got bad dns record "****************" {servfail,{dns_rec,{dns_header,12,true,query,false,false,true,true,false,2},[{dns_query,"****************",a,any,false}],[],[],[]}}`

Also in console debugger it seems : 

```
{
  "category": "uplink",
  ***"description": "Error: bad_dns",***
  "device_id": "229f4392-7ca4-4f16-a622-170ca038b7b0",
  "device_name": "",
  "id": "790b8c2b-d305-409f-8030-cbe3fa302e06",
  "inserted_at": "2022-04-20T21:52:29",
  "organization_id": "e70c2e65-d069-425f-986f-cb11eaab9b4f",
  "reported_at": "1650491549283",
  "reported_at_naive": "2022-04-20T21:52:29",
  "router_uuid": "a17d0e01-1284-4a8c-bfc5-a58f9dbe4126",
  "sub_category": "uplink_integration_res",
  "updated_at": "2022-04-20T21:52:29"
}
```

macpie, 2022-04-20:
The URL check can be very strict you can disable it via your `.env` file:

 ```
# Turn on/off HTTP integration url checks like DNS, IP... (anything else than false is on, enabled by default)
ROUTER_HTTP_CHANNEL_URL_CHECK=true
```

CanEmreUnal, 2022-04-20:
It worked! Thanks!

closed 2022-04-20 by CanEmreUnal