helium/router #712: Channel mask not sent in AU915

jamesl-dm, 2022-04-26:
**Describe the bug**
In the AU915 region, the CFList is not yet supported on Helium, so the channel mask must be sent by LinkAdrReq.
- When ADR is disabled, the LinkAdrReq to set the channel mask never happens
- When ADR is enabled, the LinkAdrReq to set the channel mask is only sent after 20 uplinks, when the ADR algorithm decides it's time to change the data rate and tx power
- If only 12.5 % of the uplinks are being received, because the channel mask isn't set, it takes 160 uplinks to trigger the LinkAdrReq

**To Reproduce**
Connect a device in AU915, and wait for the first LinkAdrReq. It will be very late if ADR is enabled, or won't be sent at all if ADR is disabled.

**Expected behavior**
- The very first uplink after the Join Request should contain a LinkAdrReq to set the channel mask, regardless of the ADR enabled / disabled setting in the profile.
- Or, adding CFList support would fix the problem for newer devices. The AU915 CFList is exactly the same format as the US915 list.

**Device Info (please complete the following information):**
 - Environment: Production
 - Console Device ID: c83da7c6-67e1-4fe3-b1c6-9b027d252028
 - Device dev eui / app eui: 0016C001F000ABA0 / 70B3D57050000009
 - Region: AU915
 - Hotspots (names) used to transfer data: perfect-chambray-panther

**Screenshots**
Screenshot for case when ADR is enabled:
[image]

mikev, 2022-05-18:
Fixed in LoRaWAN library update
https://github.com/helium/erlang-lorawan/pull/6/files

mikev, 2022-05-22:
Router updated to more recent lorawan library
https://github.com/helium/router/pull/748

closed 2022-05-23 by macpie