helium/router #719: new private console

sbittante, 2022-05-04:
1. Are you using Helium hosted Console or running your own open source Console?

> private console, OUI 25
> {"subnets":[{"mask":8388606,"base":2256}],"owner":"14ZFLcKMAeLMiMPT3vrTUNrtKm7EvNFCcqbq72uwnq3i6wWn7h6","oui":25,"nonce":3,"block":1340211,"addresses":["112kkCXnZmPCo1UonJLWRin29nS4Qcwn2J6J3ZcBLgTnWTqmRisx"]}

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).

> Organization: Redder
http://ax101.prezzibenzina.it:4000/devices/7bd980d9-ffa9-47f8-a8f7-56009992a8c9

4. Please describe your issue:

> We set up a private console few weeks ago and tested with a device (Aduenis Field Tester), however, after two weeks the device does not join anymore.
We've already run docker-compose pull and docker-compose up -d
Can you please give some hints on what logs to search?
Thanks,
Stefano

macpie, 2022-05-17:
We will need more information to be able to debug this issue, especially as you are running your own install.

Does your Router `112kkCXnZmPCo1UonJLWRin29nS4Qcwn2J6J3ZcBLgTnWTqmRisx` has any DC? It will be needed to update the XOR Filter for joins and open state channels as well to pay for data

macpie, 2022-05-17:
For what I can tell here https://explorer.helium.com/accounts/112kkCXnZmPCo1UonJLWRin29nS4Qcwn2J6J3ZcBLgTnWTqmRisx/activity your instance has not opened a state channel since April 15th. It does not have enough Data credits.

sbittante, 2022-05-17:
Hi,

yes, I update it every week by doing "docker-compose pull" and "docker-compose up -d"

according to
https://explorer.helium.com/accounts/112kkCXnZmPCo1UonJLWRin29nS4Qcwn2J6J3ZcBLgTnWTqmRisx
router 112kkCXnZmPCo1UonJLWRin29nS4Qcwn2J6J3ZcBLgTnWTqmRisx has 33,486 DC

or am I missing something?
What logs do I have to look for? What specific keywords?

Thanks,
Stefano

[image]

macpie, 2022-05-17:
`33,486 DC` is not enough to do much as state channel open ten cost 35k see previous one here https://explorer.helium.com/txns/odM83-Q5W679nChKWBP3wXx95Gy94P7eNWb8xLdduX8

sbittante, 2022-05-17:
so I was definitely missing something!

Thanks,
Stefano

macpie, 2022-05-17:
> so I was definitely missing something!
> 
> Thanks, Stefano

No worries they are few DC costs to running a Router, Xor Filter transactions to update the "join tables" and every State channel open will cost you 35k DC (or $0.35) info here https://github.com/helium/router/blob/master/.env-template#L10

closed 2022-05-17 by macpie

sbittante, 2022-05-17:
I have only one test device on the console, so I wasn't expecting to spend so may data credits....
Anyway I've just burned 1 HNT to get 800 K DC before setting
`ROUTER_SC_EXPIRATION_INTERVAL=5000`

anyway I don't understand why now I have four open state channels:
[image]