helium/router #720: AWS Integration Issue

charlesfayal, 2022-05-04:


hwahby, 2022-05-23:
Hi, @charlesfayal. Is this issue still occurring? If so, are there new device Ids? It would help troubleshoot on our side.

charlesfayal, 2022-05-24:
Honestly, it might just be best for somebody to connect to our alerts for NOWi Sensors. There's at least a few a day now that I'm connected to the alerts.
Generally, it's not a huge deal, but on multiple occasions, we've had important test nodes go down with clients watching.

hwahby, 2022-05-25:
@charlesfayal I'm happy to be added to alerts.

charlesfayal, 2022-05-27:
@hwahby okay, what's your email? I'll invite you to our org.

hwahby, 2022-05-27:
Thank you, @charlesfayal. My email is hady@nova-labs.com.

macpie, 2022-05-27:
> Thank you, @charlesfayal. My email is hady AT nova-labs DOT com

Edited for privacy.

charlesfayal, 2022-05-28:
> *Message sent from a system outside of UConn.*

Sent

closed 2022-09-07 by jdgemm