helium/router #724: AU915 Join on DR4 channel stalls the end-device

mikev, 2022-05-05:
**Describe the bug**
Browan has discovered an issue for the AU915 Region.

For AU915 the Gateway listens on a DR4 compatible channel but the Join response channel mask uses 8 to 15 and disables all DR4 compatible channels (including the one we listen on).  The device is randomly hopping around all AU915 channels and may join on any of them, but eventually is getting unlucky enough to join on this DR4 channel.

We attempted an earlier fix by setting the DRIndex and TXIndex to 0xF and 0xF.  The belief was this would be a signal to the end-device to ignore or reset it's DR settings, however the end-device continue to use DR4.  Thus the end-device is stalled.

Possible solutions the LNS side to resolve it:
* ignore uplinks on the DR4 channel
* add channel 65 or whatever the one we have to the channel mask
* when uplink on DR4 channel and we want to provide channel mask, instead of sending 0xF as datarate in LinkAdrReq, send a datarate setting that lowers the device to an acceptable level (edited)

**To Reproduce**
Discuss with browan

**Expected behavior**
End-device joins successfully

**Device Info (please complete the following information):**
 - Environment (Production ) 
 - Console Device ID
 - Device dev eui / app eui
 - How often is device sending data
 - Region AU915
 - Hotspots (names) used to transfer data - Broad-Peach-Crow

pvolodin, 2022-05-05:
I believe this issue can easily be reproduced for US915 too, seems to be a reincarnation of #633, it may be good to reopen that issue. 

> We attempted an earlier fix by setting the DRIndex and TXIndex to 0xF and 0xF. The belief was this would be a signal to the end-device to ignore or reset it's DR settings, however the end-device continue to use DR4.

Hm. #656 discussion.

pvolodin, 2022-05-05:
> Hm. #656 discussion.

What LoRaWAN version do Browan devices support?

closed 2022-06-08 by mikev