helium/router #725: EU868 TxPower

mikev, 2022-05-05:
This comment is a bit confusing to me when read with the one above.
Most of the RSSI's in the tests are changing to 30.
These two are the outliers, and I don't understand from the comments why they're both changing to 20 under opposite circumstances.

```
Above comment from L315
%% RSSI act as power here we are and based on UPLINK RSSI > -80 we should power downlink at 20

vs. this comment L457
%% RSSI act as power here we are and based on UPLINK RSSI < -80 we should power downlink at 20
```

_Originally posted by @michaeldjeffrey in https://github.com/helium/router/pull/710#discussion_r865475100_

We need to revisit the EU868 maximum tx_power setting in the LoRaWAN library.  Historically it has been set to 20, but the maximum allowed is actually 16.

mikev, 2022-06-09:
Fixed in LoRaWAN library
https://github.com/helium/erlang-lorawan/pull/13/commits/fa7b6fb30888e358671d66747091c05f507c9006

disk91, 2022-06-10:
You had a bad reading of the etsi:
- 16dbm is EIRP not DBm. 
- regulation is 14dBM for RX 1
- Regulation is 27dBm for RX 2

https://www.disk91.com/2017/technology/sigfox/all-what-you-need-to-know-about-regulation-on-rf-868mhz-for-lpwan/

mikev, 2022-06-10:
@disk91 - Can you find the exact spec which advises 14dBm for RX 1 and 27dBm for RX 2?  The only value in the spec I can find is a max value of 16dBm.

macpie, 2022-06-10:
@disk91 We have reviewed the evidence and you are right. 

Nevertheless, the regional spec makes no mention of this so we had to dig into the actual [EU standard ](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf) that you provided to see that RX2 is allowed to broadcast at `27dBM`.

As we re-wrote the regional code to  improve it, we got confused because of some previously file tickets that seemed to tell us otherwise.

- https://github.com/helium/router/issues/363
- https://github.com/helium/router/issues/364
- https://github.com/helium/router/issues/365

So we really want to make sure that: `RX1 = 14dBM ` and  `RX2 = 27dBM`. 

**Please confirm** that this is the intended behavior and will will fix it.

disk91, 2022-06-10:
Check the regulation spec : http://www.arcep.fr/fileadmin/reprise/dossiers/frequences/ERC-REC-70-03E-version02.PDF
Annex 1, Page 6 You will find the band G3 corresponding to the 869.4 to 869.65 where the RX2 windows is. This band is 10% duty cycle and 500mW e.r.p. eq 27dBm

In the LoRaWan Regional settings (RP002-1*.0.3-Final-1.pdf) you can see : 
 The RX2 receive window uses a fixed frequency and data rate. The default parameters are
 869.525 MHz / DR0 (SF12, 125 kHz)

This frequency is inside the band G3

mikev, 2022-06-10:
Fix to LoRaWAN library
https://github.com/helium/erlang-lorawan/pull/14/files

mikev, 2022-06-10:
Fix to Router
https://github.com/helium/router/pull/793

closed 2022-06-20 by mikev