helium/router #726: Console seems to not report all hotspots receiving the uplink

dmamalis, 2022-05-05:
1. Are you using Helium hosted Console or running your own open source Console?
We are using a private console hosted by Helium.

2. If you are on Helium hosted Console, which url are you currently on? 
https://console-vip.helium.com

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).
Organization: trackplast
Device ID: 9df3b703-a75e-439a-bf90-d30b16bf812a although this seems to affect multiple devices.

Please describe your issue:
There seems to be an issue with console reporting the number of hotpsots receiving each uplink. 
The Join Request is reported to be receivied by multiple hotspots as seen on the picture below. Once the Join Accept is received by the end node, then an uplink fires up within the next few seconds. This uplink is reported to be received by only one hotspot although it is transmitted in the exact same way (transmission power, SF12 etc) and the environment is exactly the same.

[image]

Another indication that the message is trully being received by multiple hotspots but the console fails to report those is that each message is reported to be seen by a different hotpsot, all of which is in the list of the initial group of hotspots which receive the Join Request.
[image]

Occasionally, the uplink message is reported to be received by two hotspots, but never more than that.
[image]

dmamalis, 2022-05-05:
@vicmgs behold live data from test-002
[image]

vicmgs, 2022-05-05:
If you click the inactive device checkbox, you can see that other hotspots are hearing the uplink but dropping them. Deferring to router team to figure out why that is happening.

dmamalis, 2022-05-06:
Is there any way those Late messages can be forwarded to the MQTT endpoint? This would **partially** solve our problem until the issue is reviewed and hopefully resolved?
This project tries to exploit reception metadata in order to geolocate devices. If the multiple packets arrive  on the console with the hotspot name and timestamp, then this would bring back some confidence to the whole project.

[image]

macpie, 2022-05-09:
Seems like you are trying to `ack` on every message which make the window on time small (so we are able to downlink). If you don't care too much about the `ack` I would stop so that your window of time can be bigger and you should get more packets in there.

closed 2022-05-09 by macpie

dmamalis, 2022-05-10:
@macpie first things first the image is from a test device which does send a lot of messages. The normal operation of the rest of the devices is to send a single message every 6 hours. The message is indeed a confirmed uplink with a LinkCheck request. If the device has adequate coverage it will drop it's TX power, this is an extra mechanism on top of the ADR.

No matter what, asking for a downlink is a normal process in LoRaWAN. As I understand, latency issues delay this functionality. This is something we can live with, knowing that the light hotspots will eventually make things better.

However, as I aforementioned, since the router delivers the late messages on the console, it would be helpful to deliver those messages in all endpoints (etc MQTT integration) including the metadata. This would allow application owners to investigate the data or draw other types of useful information from that data.

In our case, we switched to Helium, due to its vast coverage, in order to run trilateration algorithms and locate the devices but this behavior is a deal breaker for the time being. If the router can send Late Packets to an integration endpoint then the problem is minimized. I do think that this will be a vast improvement in the functionality and reliability of the network. Late or not, these packets do carry a lot more information than just the data :-)

macpie, 2022-05-11:
> @macpie first things first the image is from a test device which does send a lot of messages. The normal operation of the rest of the devices is to send a single message every 6 hours. The message [...]

It is an interesting feature, would you mind create a new ticket for the feature? (Deliver late packets)

dmamalis, 2022-05-11:
> It is an interesting feature, would you mind create a new ticket for the feature? (Deliver late packets)

I do not mind at all @macpie but I do mind for the data asap as this commercial project is actually crumbling as we speak