helium/router #740: Add Support for Regions IN865 and KR920

mikev, 2022-05-16:
The Helium Network currently lacks support for the IN865 and KR920 frequency region plans.  With this ticket we'll update and QA these plans so that Hotspot owners in India, Pakistan and South Korea can use the Helium network.

Fully supporting a new Region frequency plan requires coordination and support from the Router, Gateway-RS, H3, firmware and various other projects.  Finally it also requires some level of software and hardware QA testing to validate that real IN868 end-devices can physically connect and Join to an IN868 Hotspot.

mikev, 2022-05-18:
LoRaWAN library update
https://github.com/helium/erlang-lorawan/pull/6/files

mikev, 2022-05-22:
Library changes merged to master
https://github.com/helium/router/pull/748

jdgemm, 2022-06-01:
duplicated, by #771, #762

closed 2022-06-01 by jdgemm