helium/router #742: Azure IoT Hub   "Error: bad_dns" and "Error: not_connected"

EnrSal, 2022-05-16:
**Describe the bug**
Azure IoT Hub not connecting in Integration. This started in the last couple of days.

The integration message is as follows:
{
  "category": "uplink",
  "description": "Error: bad_dns",
  "device_id": "3520f2ac-196f-4516-a394-80a80627eff2",
  "device_name": "StartMikeAgain",
  "id": "a8054c0f-e9dc-47bc-a177-8091fe49a6e3",
  "inserted_at": "2022-05-16T22:48:00",
  "organization_id": "7d6ef6d5-5da3-4b39-a1ec-6929aff512dc",
  "reported_at": "1652741279999",
  "reported_at_naive": "2022-05-16T22:47:59",
  "router_uuid": "fcd56f09-0263-4bd2-a64d-5f32c0ce79be",
  "sub_category": "uplink_integration_res",
  "updated_at": "2022-05-16T22:48:00"
}

  "category": "uplink",
  "description": "Error: not_connected",
  "device_id": "3520f2ac-196f-4516-a394-80a80627eff2",
  "device_name": "StartMikeAgain",
  "id": "61b41681-7c03-46e8-b2b4-71e4dc602a28",
  "inserted_at": "2022-05-16T22:48:00",
  "organization_id": "7d6ef6d5-5da3-4b39-a1ec-6929aff512dc",
  "reported_at": "1652741280000",
  "reported_at_naive": "2022-05-16T22:48:00",
  "router_uuid": "fcd56f09-0263-4bd2-a64d-5f32c0ce79be",
  "sub_category": "uplink_integration_res",
  "updated_at": "2022-05-16T22:48:00"
}

**To Reproduce**
Send to Azure IOT hub

**Expected behavior**
This has been working since 2 months ago after it was fixed after issue #634 was resolved.
The bad DNS error was not part of #634.

**Device Info (please complete the following information):**
  - production
 - Console Device ID
 - 3520f2ac-196f-4516-a394-80a80627eff2

 every 3 minutes and then every 15 minutes

EnrSal, 2022-05-17:
Something changed today.
I am getting data to the Azure IoT hub but I still see "Error: bad_dns".

{
  "category": "uplink",
  "description": "Error: bad_dns",
  "device_id": "3520f2ac-196f-4516-a394-80a80627eff2",
  "device_name": "StartMikeAgain",
  "id": "35d698d0-c531-4177-95ab-f004187d2f1a",
  "inserted_at": "2022-05-17T17:30:15",
  "organization_id": "7d6ef6d5-5da3-4b39-a1ec-6929aff512dc",
  "reported_at": "1652808615642",
  "reported_at_naive": "2022-05-17T17:30:15",
  "router_uuid": "d856d689-3259-44e1-b9d7-6cde8199bb16",
  "sub_category": "uplink_integration_req",
  "updated_at": "2022-05-17T17:30:15"
}

michaeldjeffrey, 2022-05-17:
Looks like your device is hooked up to 2 integrations.
The iot-hub integration looks to be working with the token you provided.
Can you check on the other integration to make sure everything is setup correctly?

EnrSal, 2022-05-17:
I need to have a downlink URL to properly get my queued   commands.
Is this related to:
https://github.com/helium/router/issues/744

Thanks

EnrSal, 2022-05-17:
Is this tutorial still relevant?
This is the way I did my integration.
https://docs.helium.com/use-the-network/console/integrations/http/

michaeldjeffrey, 2022-05-17:
> I need to have a downlink URL to properly get my queued commands. 
> Is this related to: #744 
> Thanks

That issue is from someone running their own Console/Router instance. 
The Helium Foundation Console provides a properly valid downlink_url.

> Is this tutorial still relevant?
> This is the way I did my integration.
> https://docs.helium.com/use-the-network/console/integrations/http/

That docs page is still valid. 
You have a malformed URL in your integration.

closed 2022-05-17 by michaeldjeffrey