helium/router #744: Downlink URL invalid in json message

disk91, 2022-05-17:
**Describe the bug**
The donwlink_url field is not correct on the integration json payload for the open-source console.
It doesn't contains the server URL but just the path, even if on the integration page on the console, the full url is correctly displayed.

```
Integration
Debug Request
{
  "id": "bb92",
  "name": "WioFieldTesterIntegration",
  "status": "success"
}
{
  "body": {
    "app_eui": "",
    "dc": {
      "balance": 278,
      "nonce": 1
    },
    "dev_eui": "",
    "devaddr": "",
    "downlink_url": "/api/v1/down/bede81e5-2b92/JsUihIcJWptTjrgPFy/6f73f6e17c518",
    "fcnt": 0,
    "hotspots": [
      {

```

**To Reproduce**
Steps to reproduce the behavior.
Create an integration on console.helium-iot.eu and watch the integration downlink vs the debug downlink_url reality

**Expected behavior**
I expect to get `"downlink_url": "https://console.helium-iot.eu/api/v1/down/bede81e5-2b92/JsUihIcJWptTjrgPFy/6f73f6e17c518",` as in the nova console

michaeldjeffrey, 2022-05-17:
Is this what you're looking for?
https://github.com/helium/router/blob/master/.env-template#L26-L27

```
# Public facing endpoint to POST downlinks for devices
ROUTER_CONSOLE_DOWNLINK_ENDPOINT=http://helium_console:4000
```

disk91, 2022-05-18:
The router template file contains it but the console router-template does not contain it. 
both should be aligned to avoir such situation.

macpie, 2022-06-07:
Related PR https://github.com/helium/console/pull/1187

closed 2022-06-07 by macpie