helium/router #749: Uplink data received on Syncrobit hotspot not received on console

denicola2, 2022-05-20:
**Describe the bug**
Uplink data confirmed to be received by the hotspot (as seen in packet forwarding logs) is not seen on the console (console.helium.com). This is happening for all devices onboarded to my account which were previously working prior to the light hotspot rollout. Many of such devices stopped functioning on May 15, while others stopped at May 18. Since then, no uplinks have reached the console.

**To Reproduce**
Trigger any uplink. In my case using a mapper to manually trigger uplink.
Witness the uplink packet in the packet forwarding logs as seen here:
```
INFO: [up] PUSH_ACK received in 2 ms
JSON up: {"rxpk":[{"jver":1,"tmst":606883478,"chan":7,"rfch":1,"freq":905.300000,"mid": 8,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","rssis":-62,"lsnr":13.8,"foff":5462,"rssi":-61,"size":24,"data":"QL0BAEgAwVICZryBeNqToZrgaSO3DbfN"}]}

INFO: Received pkt from mote: 480001BD (fcnt=21185)

INFO: [up] PUSH_ACK received in 1 ms
JSON up: {"stat":{"time":"2022-05-19 16:17:47 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0,"temp":30.0}}
```
Additional logs from forwarder:
```
##### END #####
### Concentrator temperature: 30 C ###
# GPS sync is disabled
### [GPS] ###
src/jitqueue.c:442:jit_print_queue(): INFO: [jit] queue is empty
#--------
src/jitqueue.c:442:jit_print_queue(): INFO: [jit] queue is empty
### [JIT] ###
# BEACON rejected: 0
# BEACON sent so far: 0
# BEACON queued: 0
# SX1302 counter (PPS):  0
# SX1302 counter (INST): 2254638905
### SX1302 Status ###
# TX errors: 0
# RF packets sent to concentrator: 0 (0 bytes)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# PULL_DATA sent: 3 (100.00% acknowledged)
### [DOWNSTREAM] ###
# PUSH_DATA acknowledged: 100.00%
# PUSH_DATA datagrams sent: 4 (872 bytes)
# RF packets forwarded: 3 (72 bytes)
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets received by concentrator: 3
### [UPSTREAM] ###
##### 2022-05-19 16:45:18 GMT #####
```
Additional logs from miner:
```
2022-05-19 17:06:40.832 1810 [info] <0.1854.0>@miner_lora_light:handle_udp_packet:350 PULL_DATA from 12273815315514654720 on 58127
2022-05-19 17:06:38.169 1810 [info] <0.1826.0>@blockchain_state_channels_client:handle_packet:381 handle_packet #packet_pb{oui=0,type=lorawan,payload=<<64,189,1,0,72,0,200,82,2,108,179,66,152,24,190,103,82,187,37,133,151,25,143,46>>,timestamp=239261774,signal_strength=-69,frequency=905.1,datarate=<<"SF7BW125">>,snr=13.2,routing=#routing_information_pb{data={devaddr,1207959997}},rx2_window=undefined} to ["OUI 1"] with handler blockchain_state_channel_handler
```
Observe the debugging panel on console for said mapper device and see no uplinks present. 
Observe no uplinks present in the event log as well.
All DevEUP, AppEUI, and AppKey have all been verified, especially as this mapper was working prior to the 15th.
Similar behavior is seen with all sensors in my home where my Syncrobit hotspot is the sole provider of lorawan coverage. All devices have had prior successful uplinks.

**Expected behavior**
Observe JOIN and UPLINK packets on the console
Observe UPLINK packets forwarded to mappers and cargo integrations.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 Production
 - Console Device ID
e14504a4-bbbe-46bb-8df8-fbb683e32e3e
 - Device dev eui / app eui
 30C6F71EFEACFEFF / 6081F9BF908E2EA0
 - How often is device sending data
 Manually triggering upload several times at different intervals.
 - Region
 US915
 - Hotspots (names) used to transfer data
 lone-coral-raccoon
- Additional Device information
mapper built on TTGO TBeam using MaxPlastix latest release
(https://github.com/Max-Plastix/tbeam-helium-mapper)

*NOTE: do not post your device `app_key`*

**Additional context**
Other devices currently encountering the issue:
Dragino LDS02: 
aa128fca-d0d8-43d1-a6d8-7dbf6a6a525f
A840413DB1830A7D
A000000000000107

Dragino LDS02:
9ba0af90-e8c2-47d8-8f3d-18c0b2467734
A84041AEA1846253
A840410000000107

Dragino LDDS20:
eab2a8f4-d79d-4c00-a1bf-f82a16aff135
A840411AC18436CF
A840410000000101

RAK Wisblock:
e2a015f8-9641-41fa-9936-6de778c7bf95
AC1F09FFFE06D394
6081F9EEDA97B8C6

**Screenshots**
Empty Event log:
[image]

denicola2, 2022-05-20:
**Additional hotspot information**
Firmware Version: 2022.05.19.0
Miner Name: lone-coral-raccoon
Miner Address: 112U4oPUekp2SNetqixKGuw6EHU3tRCuvyrpfYFrwKxmLMMpL4Pd
Region: US915
Concentrator Model: sx1302

michaeldjeffrey, 2022-05-20:
Can you try rejoining this device `e14504a4-bbbe-46bb-8df8-fbb683e32e3e`?
It doesn't look to have a devaddr in Router.

denicola2, 2022-05-20:
I have tried both:
- Recreating the device fully in console
- Triggering join by manually resetting the mapper

It seems the join is likewise not reaching past the hotspot.

michaeldjeffrey, 2022-05-20:
Recreating the device in console will wipe everything we know about it, and the device will have to rejoin.
I asked over in Discord how you got a devaddr. 
Did you recreate the device and _not_ rejoin?
If that's the case, Router can't know to buy the packet, because it doesn't have a session with your device.

denicola2, 2022-05-20:
The info I have given here is the correct info since I've begin testing today. I have triggered re-join multiple times since (with the info provided above).

michaeldjeffrey, 2022-05-20:
I've added a trace for this device. 
Can you trying joining again, and maybe sending some more packets?

Unfortunately, I won't be able to look more into this until tomorrow morning.

denicola2, 2022-05-20:
Additional uplinks sent:
```
INFO: [up] PUSH_ACK received in 19 ms
JSON up: {"rxpk":[{"jver":1,"tmst":3789412301,"chan":0,"rfch":0,"freq":903.900000,"mid": 8,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","rssis":-75,"lsnr":13.0,"foff":4600,"rssi":-75,"size":24,"data":"QL0BAEgA0VICYbhoKLQyLwhh6MMBHXTg"}]}

INFO: Received pkt from mote: 480001BD (fcnt=21201)

INFO: [up] PUSH_ACK received in 1 ms
JSON up: {"rxpk":[{"jver":1,"tmst":3786785139,"chan":7,"rfch":1,"freq":905.300000,"mid": 8,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","rssis":-65,"lsnr":14.2,"foff":4272,"rssi":-65,"size":24,"data":"QL0BAEgA0FIC+Oy5SXnizFm8cKLHyDe/"}]}

INFO: Received pkt from mote: 480001BD (fcnt=21200)

INFO: [down] PULL_ACK received in 0 ms
INFO: [down] PULL_ACK received in 0 ms
INFO: [up] PUSH_ACK received in 1 ms
JSON up: {"stat":{"time":"2022-05-20 04:10:42 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0,"temp":30.0}}
```

denicola2, 2022-05-20:
Additional logs from miner. Not sure if relevant.
```
2022-05-20 04:14:12.382 1671 [info] <0.1889.0>@miner_lora_light:handle_json_data:440 Gateway #gateway{mac=12273815315514654720,ip={127,0,0,1},port=57405,sent=0,received=2244,dropped=0,status=#{<<"ackr">> => 100.0,<<"dwnb">> => 0,<<"rxfw">> => 0,<<"rxnb">> => 0,<<"rxok">> => 0,<<"temp">> => 30.0,<<"time">> => <<"2022-05-20 04:14:12 GMT">>,<<"txnb">> => 0},rtt_samples=[],rtt=5000000}
2022-05-20 04:14:12.381 1671 [info] <0.1889.0>@miner_lora_light:handle_json_data:439 got status #{<<"ackr">> => 100.0,<<"dwnb">> => 0,<<"rxfw">> => 0,<<"rxnb">> => 0,<<"rxok">> => 0,<<"temp">> => 30.0,<<"time">> => <<"2022-05-20 04:14:12 GMT">>,<<"txnb">> => 0}
2022-05-20 04:14:12.381 1671 [info] <0.1889.0>@miner_lora_light:handle_udp_packet:326 PUSH_DATA [{<<"stat">>,[{<<"time">>,<<"2022-05-20 04:14:12 GMT">>},{<<"rxnb">>,0},{<<"rxok">>,0},{<<"rxfw">>,0},{<<"ackr">>,100.0},{<<"dwnb">>,0},{<<"txnb">>,0},{<<"temp">>,30.0}]}] from 12273815315514654720 on 52879
```

denicola2, 2022-05-20:
Hotspot has supposedly transferred a packet, though it doesnt seem to belong to any of my devices
[image]

satonakamo, 2022-05-20:
I have the same problem with a different hotspot maker (Pisces P100) but with the same sensor (TTGO TBeam using MaxPlastix latest release). In my case, in the console the data is reported by a much distant hotspot with a much lower SNR. Package forward analyzer confirms the data is picked up by Lora and forwarded to the miner. Here is my console log output:
``` 
2022-05-20 11:29:05.489 7 [info] <0.21374.4>@blockchain_state_channels_client:handle_packet:{381,5} handle_packet #packet_pb{oui=0,type=lorawan,payload=<<64,99,3,0,72,0,1,1,2,215,254,189,233,152,12,212,131,127,165,144,27,246,231,222>>,timestamp=2664015882,signal_strength=-48,frequency=868.5,datarate=<<"SF7BW125">>,snr=11.5,routing=#routing_information_pb{data={devaddr,1207960419}},rx2_window=undefined} to ["OUI 1"] with handler blockchain_state_channel_handler
```

gosecure-rseguin, 2022-05-20:
Same issue with an Dragino LP8... Packets registering with the hotspot, but not shown on the dashboard

macpie, 2022-05-20:
2 things here:

1. You can take down your logs to debug via `/sbin/miner eval 'lager:set_loglevel(lager_file_backend, "console.log", debug).'` This would get us more information.
2. Some hotspots are having issue because of latest update.

denicola2, 2022-05-21:
Unsure about the procedure for getting the requested logs. Got the previous logs via the Syncrobit local dashboard.

satonakamo, 2022-05-21:
>     1. You can take down your logs to debug via `/sbin/miner eval 'lager:set_loglevel(lager_file_backend, "console.log", debug).'` 
Can you be more explicit? There is no such file on my miner so I am getting errors when I issue the command. The device is a Pisces P100 running `miner-arm64_2022.05.19.0_GA` firmware

denicola2, 2022-05-22:
I have confirmed with the mapper mentioned in the original bug that other neighboring hotspots have successfully forwarded my device's uplinks.

denicola2, 2022-05-23:
Now going on a week that my devices' data is not seen in the console. Hopefully there are not too many other enterprise-level customers that are encountering similar issues.

macpie, 2022-05-23:
The hotspot has transferred data just a day ago [see activity](https://explorer.helium.com/hotspots/112U4oPUekp2SNetqixKGuw6EHU3tRCuvyrpfYFrwKxmLMMpL4Pd/activity). So I think we can eliminate the hotspot being an issue. Since you have deleted the device would you mind providing the new device ID?

_Note: we will only be looking at @denicola2 's issue otherwise this becomes very confusing_

denicola2, 2022-05-23:
As mentioned here (https://github.com/helium/router/issues/749#issuecomment-1132349444) the device ID provided (e14504a4-bbbe-46bb-8df8-fbb683e32e3e) is the one I have been testing with. The device deletion and re-addition testing was done prior to opening the git issue. Just wanted to mention that it was something I had tried as a matter of process of elimination. 

Also mentioned above, that same device (e14504a4-bbbe-46bb-8df8-fbb683e32e3e) was able to successfully join and transmit data via neighboring hotspots. Uplinks stopped when in an area where coverage was provided by only my hotspot.

macpie, 2022-05-23:
> As mentioned here ([#749 (comment)](https://github.com/helium/router/issues/749#issuecomment-1132349444)) the device ID provided (e14504a4-bbbe-46bb-8df8-fbb683e32e3e) is the one I have been testing with. The device deletion and re-addition testing was done prior to opening the git issue. Just wanted to mention that it was something I had tried as a matter of process of elimination.
> 
> Also mentioned above, that same device (e14504a4-bbbe-46bb-8df8-fbb683e32e3e) was able to successfully join and transmit data via neighboring hotspots. Uplinks stopped when in an area where coverage was provided by only my hotspot.

Thanks, it seems that we are having an issue with our latest update to our LoraWan implementation. We are working on reverting changes and releasing again asap.

denicola2, 2022-05-23:
Confirming that uplinks are now functional again after the update to v2022.05.12.8

closed 2022-05-23 by denicola2

macpie, 2022-05-23:
> Confirming that uplinks are now functional again after the update to v2022.05.12.8

Great thank you!