helium/router #753: Join accept sending in wrong channel

CanEmreUnal, 2022-05-22:
Device tries to connect to hotspot within US915 freq channel but join accept is sending back by hotspot to the device in EU868.

Packet-forwarder conf seems fine. 

Hotspot is located in EU.

- What kind of mechanism behind is selecting the write frequency plan ?
- Could moving the hotspot to the US might be fix it ?

```
[Logs]    [2022-5-23 1:17:43] [packet-forwarder] JSON down: {"txpk":{"codr":"4/5","data":"ILvNqFA9lCMJWaQRXf/xc28=","datr":"SF9BW125","freq":867.2999877929688,"imme":false,"ipol":true,"modu":"LORA","powe":20,"rfch":0,"size":17,"tmst":1784173964}}
[Logs]    [2022-5-23 1:17:43] [packet-forwarder] ERROR: Packet REJECTED, unsupported frequency - 867299987 (min:902000000,max:928000000)
```

[image]

macpie, 2022-05-23:
Was this fixed with `2022.05.12.8`?

CanEmreUnal, 2022-05-26:
@macpie it is fixed by updating location of the hotspot to USA zone.

closed 2022-05-26 by CanEmreUnal

macpie, 2022-05-26:
> @macpie it is fixed by updating location of the hotspot to USA zone.

Thx for the update