helium/router #763: Uplink packets are dropped as late, with repeating frame counts and low hold_times

satonakamo, 2022-05-24:
**Describe the bug**
Running a TTGO T-beam [1] device with mapper and cargo integration on the console. Observing that uplink packet from my Pisces P100 hotspot (miner-arm64_2022.05.19.0_GA ) are constantly being dropped due to being too late. 
During tests, when the device is a few meters away from the hotspot, the console is often reporting more than one packet with the same frame count (eg 541) from my hotspot being dropped with low `hold_times` (~500ms). The only way I am able to force some data transmit is by selecting only this hotspot as preferred (that's why you can see data transmits on the explorer). Even then there are dropped packets.

[image]

Attached are the console logs and the packet forwarder logs of the hotspot.
[packet_forwarder.log](https://github.com/helium/router/files/8762210/packet_forwarder.log)
[event-debug.log](https://github.com/helium/router/files/8762231/event-debug.log)

[1] https://github.com/Max-Plastix/tbeam-helium-mapper

**Expected behavior**
I would like to map the coverage of my hotspot. I am unable to do so because packets are dropped. The issue is only affecting my hotspot, nearby hotspots are transmitting the mapper packets in time.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 production on console.helium.com
 - Console Device ID
 bd0d36fe-3c0a-442d-bb57-6a4b81e187bb
 - Device dev eui / app eui
dev: 6081F90C9675423D
app: 6081F956A22E69A6
 - How often is device sending data
~every minute, or on demand
 - Region
 EU
 - Hotspots (names) used to transfer data
 Trendy Emerald Cheetah
112RqZwoKsRH97CufLXhVomzrNviHpPgQEYzvCBEvvwh9QWc5skR

satonakamo, 2022-05-27:
Can confirm now that the bug was caused due to the hotspot.
It has a build in GPS sensor and once I removed the following lines from the packet forwarders `global_conf.json` the packets stopped being delayed. 
```"gps_tty_path": "/dev/serial0"```
I suspect leaving this line in caused the GPS timing to be used. Since most hotspots don't have GPS this hotspot was becoming out of sync. This was also evident in the witness reports, where this hotpot was also consistently later than others. Delayed witness reports are still valid so it could be that there are several other hotspots affected by this issue, but are not aware that they are missing data packets.

closed 2022-05-27 by satonakamo

macpie, 2022-05-27:
Thanks for the update!