helium/router #781: Multibuy having issues with downlink que

r4wk, 2022-06-04:
1. Are you using Helium hosted Console or running your own open source Console? Helium

2. If you are on Helium hosted Console, which url are you currently on? N/A

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753). r4wknet, b49366bf-b8fe-4c03-900f-444ef87b4c78

Please describe your issue:
When I have a downlink queued in the console it seems like multibuy doesn't work as intended. As in the console will only report one hot spot has heard the uplink (in a very rare occasion I've seen 2 hot spots). As soon as I completely clear the queue my uplinks are heard by multiple hot spots. Here is very small example. You can see the exact uplink where I stopped all downlinks (6:08:53). I can queue up a bunch of downlinks and I'll go back to only displaying 1 hot spot (the one that handles the downlink), but as soon as the downlink queue is cleared it displays number of hot spots correctly.

TL;DR When there's no downlink queued it is very consistently displaying 2-4 hot spots in my test location. As soon as I queue up downlinks it's only displaying 1. Please let me know if you need anymore info! 

[image]

macpie, 2022-06-06:
Yes, you are right and this is due to the fact that to respect potential low `Rx delay` (window of time Router has to respond to device).
If Router knows there is a downlink queued it will not wait as long as if it had nothing in queue (we are talking few milliseconds when a downlink is queued compared to potentially fews seconds when none). So it gives more time for more hotspot to report their copies.

We have (not too long ago) implemented the option to set your `rx_delay` but we never really included it in the calculation in the "waiting time". We will use this ticket as a place holder for this "new feature".

Feature should allow user the set their own "waiting time" so Router could wait longer to gather more copies of the packets (so more hotspot) while still respecting `rx_delay` values.

r4wk, 2022-06-06:
Ah ha! Good to know, thank you for the explanation. Makes total sense now. 

Glad it's not a bug and more of a feature request. You all are doing a great job.

closed 2022-09-07 by jdgemm

r4wk, 2022-09-14:
Was this ever added? I'm guessing most feature requests have been dropped with the possibility of moving to Chirpstack?