helium/router #785: Device join failed using Helium hosted Console

kanalcl97, 2022-06-08:
1.Are you using Helium hosted Console or running your own open source Console?
 >I am using Helium hosted Console.

2.If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).

    Organization: sercomm_test
    Device ID: c93ca022-c034-4d43-9afc-42fe9be1c2c1

3.Please describe your issue:
>We are using Helium Hosted console to test a lora device and it worked fine until 27 of May, and it cannot join successfully when we tested yesterday(7th of June).
    We can see Join Request & Join Accept from Helium Console but Lora device prints "Join Failed".
    The dataonly hotspot we run is pulled from gateway-rs source code.  we can see downlink packet sending out from helium_gateway. 
    We did not make any changes in configurations to dataonly hotspot and lora gateway but the device just cannot join.
    Can you please tell us how to fix this or is there any change in Helium Hosted Console might lead to this issue?
    Thanks,
    Kana

macpie, 2022-06-08:
Could you provide the Hotspot being used?

kanalcl97, 2022-06-09:
The Hotspot is 13ZTnbGwMhTo8uijjrJgRNQXCBmZeiJMMrs4ZxVLYuFCuLkqkSs named Flat Opaque Platpus.
It is a dataonly hotspot.