helium/router #795: MQTT Integration Problem

CanEmreUnal, 2022-06-14:
I've set up the mqtt integration for a few devices, which was working fine for 2 days.

Today I'm facing the following logs while the http integration is working fine.

Possible remote server connection drops are causing this state, but I need to step into it without restarting the container.

BACKOFF strategy is not proper it seems.

https://github.com/helium/router/blob/26e3e3463ec0b12dac3313e16837ae26015c5e46/src/channels/router_mqtt_channel.erl

`[2022-06-14 17:27:09.583] <0.14018.0> [error] [8d82c718-861b-4957-b2d1-e918f14a30e4] [router_mqtt_channel:publish:{346,13}] [54cb6c6f-c0ac-4e91-8a47-3d24494eaeca] failed to publish`

vkynchev, 2022-08-29:
I can confirm the issue. On random intervals some device receive Integration response and their connections are shown as 'Disconnected' on broker side.  Reconnection is made after I modify the integration URL (for example change port and then revert back).

closed 2022-09-07 by jdgemm

vkynchev, 2022-09-11:
@jdgemm why close the issue? It’s still relevant…

[image]